lgsvl/simulator #1445: Camera projections inconsistencies

jacoblambert, 2021-05-18:
## Description of the issue
I'm trying to test a checkerboard corner detector. The idea is simple: with a known camera (position, orientation, intrinsic parameters and distortions) and board (position, orientation), I can obtain the 3D position of the checkerboard corners in the camera frame, then project them in the image frame to obtain ground truth 2D pixel position of the chessboard corners. I'm using the Python API to do this.

So I've obtain a 3D chessboard model made in blender. It is 8 squares in width and 6 squares in height with a square size of 0.1 meters. I can therefore easily produce a list of 3D points representing the board at the origin.
[image]
```
def generate_checkerboard_3d_points():
        board_points = []
        for i in np.linspace(-8 * 0.1 / 2, 8 * 0.1 / 2, 8+1):
            for j in np.linspace(-6 * 0.1 / 2, 6 * 0.1 / 2):
                board_points.append(np.array([i,j,0])
        return np.array(board_points, dtype=np.float64)
```
The results is a checkerboard at the origin, including outer corners.
```
board_points
array([[-0.4, -0.3,  0. ],
       [-0.4, -0.2,  0. ],
       [-0.4, -0.1,  0. ],
       [-0.4,  0. ,  0. ],
       [-0.4,  0.1,  0. ],
       [-0.4,  0.2,  0. ],
       [-0.4,  0.3,  0. ],
       [-0.3, -0.3,  0. ],
       [-0.3, -0.2,  0. ],
       [-0.3, -0.1,  0. ],
       [-0.3,  0. ,  0. ],
       [-0.3,  0.1,  0. ],
       [-0.3,  0.2,  0. ],
       [-0.3,  0.3,  0. ],
       [-0.2, -0.3,  0. ],
       [-0.2, -0.2,  0. ],
       [-0.2, -0.1,  0. ],
       [-0.2,  0. ,  0. ],
       [-0.2,  0.1,  0. ],
       [-0.2,  0.2,  0. ],
       [-0.2,  0.3,  0. ],
       [-0.1, -0.3,  0. ],
       [-0.1, -0.2,  0. ],
       [-0.1, -0.1,  0. ],
       [-0.1,  0. ,  0. ],
       [-0.1,  0.1,  0. ],
       [-0.1,  0.2,  0. ],
       [-0.1,  0.3,  0. ],
       [ 0. , -0.3,  0. ],
       [ 0. , -0.2,  0. ],
       [ 0. , -0.1,  0. ],
       [ 0. ,  0. ,  0. ],
       [ 0. ,  0.1,  0. ],
       [ 0. ,  0.2,  0. ],
       [ 0. ,  0.3,  0. ],
       [ 0.1, -0.3,  0. ],
       [ 0.1, -0.2,  0. ],
       [ 0.1, -0.1,  0. ],
       [ 0.1,  0. ,  0. ],
       [ 0.1,  0.1,  0. ],
       [ 0.1,  0.2,  0. ],
       [ 0.1,  0.3,  0. ],
       [ 0.2, -0.3,  0. ],
       [ 0.2, -0.2,  0. ],
       [ 0.2, -0.1,  0. ],
       [ 0.2,  0. ,  0. ],
       [ 0.2,  0.1,  0. ],
       [ 0.2,  0.2,  0. ],
       [ 0.2,  0.3,  0. ],
       [ 0.3, -0.3,  0. ],
       [ 0.3, -0.2,  0. ],
       [ 0.3, -0.1,  0. ],
       [ 0.3,  0. ,  0. ],
       [ 0.3,  0.1,  0. ],
       [ 0.3,  0.2,  0. ],
       [ 0.3,  0.3,  0. ],
       [ 0.4, -0.3,  0. ],
       [ 0.4, -0.2,  0. ],
       [ 0.4, -0.1,  0. ],
       [ 0.4,  0. ,  0. ],
       [ 0.4,  0.1,  0. ],
       [ 0.4,  0.2,  0. ],
       [ 0.4,  0.3,  0. ]])
```
I define two boards at the following positions:
```
board_positions = []
# Right board
board_positions.append({'position': lgsvl.Vector(0.6, 0.1, 4.0),
                        'rotation': lgsvl.Vector(0, 0, 0)})
# Left board
board_positions.append({'position': lgsvl.Vector(-0.8, 1.0, 4.5),
                        'rotation': lgsvl.Vector(-50, -80, 0)})
```
And get this kind of scene:
[image]

I define the camera to be slightly above and to the front of the vehicle. I set distortions to false initially.
```
  {
    "type": "Color Camera",
    "name": "RoofCamera",
    "params": {
      "Width": 2880,
      "Height": 1860,
      "Frequency": 1,
      "JpegQuality": 100,
      "FieldOfView": 38.58,
      "MinDistance": 0.1,
      "MaxDistance": 100,
      "Topic": "/front_camera/image_raw",
      "Frame": "front_camera",
      "Distorted": false,
      "DistortionParameters": [-0.2159,    0.2200, -0.1573, 0]
    },
    "transform": {
      "x": 0,
      "y": 1,
      "z": 2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  }
```
And obtain this image.
[image]
To be sure, I obtain image size and vertical FOV from the simulator, and calculate the camera intrinsics:
```
image_width = sensor_camera.width
image_height = sensor_camera.height
aspect_ratio = image_width / image_height
vertical_fov = sensor_camera.fov
horizon_fov = 2 * math.degrees(math.atan(math.tan(math.radians(vertical_fov) / 2) * aspect_ratio))
fx = image_width / (2 * math.tan(0.5 * math.radians(horizon_fov)))
fy = image_height / (2 * math.tan(0.5 * math.radians(vertical_fov)))
cx = image_width / 2
cy = image_height / 2
```
and obtain the follow camera matrix:
```
>>> camera_intrinsics
array([[2657.1496, 0, 1440],
       [0, 2657.1496, 930],
       [0, 0, 1])
```
To obtain the 3D point in the camera frame, I get the camera position and board positions directly from the simulator, and get their transformation matrix using LGSVL's transform_to_matrix function.
```
board_world_tf = np.array(transform_to_matrix(lgsvl.Transform(location, rotation)))
camera_world_tf = np.array(transform_to_matrix(self.camera_sensor.transform)) 
```
```
# Right board
>>>board_world_tf
array([[ 1.        ,  0.        ,  0.        ,  0.        ],
       [ 0.        ,  1.        ,  0.        ,  0.        ],
       [ 0.        , -0.        ,  1.        ,  0.        ],
       [ 0.60000002,  1.        ,  4.        ,  1.        ]])
# Left board
>>> board_world_tf
array([[ 1.73648143e-01, -2.98023212e-08,  9.84807759e-01, 0],
       [ 7.54406515e-01,  6.42787610e-01, -1.33022176e-01, 0],
       [-6.33022222e-01,  7.66044443e-01,  1.11618897e-01, 0,
       [-8.00000012e-01,  1.00000000e+00,  4.50000000e+00, 1]])
>>> camera_world_tf
array([[ 1.,  0.,  0.,  0.],
       [ 0.,  1.,  0.,  0.],
       [ 0., -0.,  1.,  0.],
       [ 0.,  1.,  2.,  1.]])

```
As I understand, these represent the board to world TF and camera to world TF respectively. By inverting the camera_world_tf I therefore get: points in board space -> points in world space -> points in camera space as so:
```
checkerboard_points_h = np.ones((checkerboard_points.shape[0], 4), dtype=np.float64)
checkerboard_points_h[:, :3] = checkerboard_points
# TF
corners_3D = np.dot(np.dot(checkerboard_points_h, board_world_tf), np.linalg.inv(camera_world_tf))

corners_3D = corners_3D / corners_3D[:, 3].reshape(corners_3D.shape[0], 1)
corners_3D = corners_3D[:, :3]
```
Finally, I use OpenCV to project the points the camera frame to the image frame, with all distortion coefficients set to 0.
```
board_points_2d, _ = cv2.projectPoints(board_3d_points,
      np.zeros((1,3), dtype=np.float64),
      np.zeros((1,3), dtype=np.float64),
      camera_intrinsics,
      distortion_coefficients
      )
```
I obtain the following, which seems close but incorrect. Not sure why as this TF chain seems correct, but perhaps Unity's left handed coordinate system not coordinating with OpenCV. Flipping the y axis of the 3D corners _after_ the transformation fixes the issue.
[image]
`corners_3D[:, 1] = -corners_3D[:, 1]`
[image]
This _looks_ correct at a glance, but its not, there's an offset that is difficult to explain when the corners really should match exactly.
[image]
[image]

If I include distortion parameters, this gets worse:
[image]

## What I Tried
* I've tried different approaches to transform my points from 3D board space to 3D camera space (not a transform issue?)
* I re-implemented the 3D to 2D projection and the result is the same as using opencv (not an opencv compatibility issue?)
* I've re-implemented OpenCV's distortion model which matches the LGSVL distortion model as far as I can see, minus the tangential distortion. If I obtain a distorted image from LGSVL or obtain an undistorted image from LGSVL and distort it with OpenCV, the result is the same, at least visually.

## Conclusion
Clearly there is something wrong with my projections but I can't figure out what. Perhaps some silly bug, but it seems odd that without distortion the model is _slightly_ off. The only thing I can think of is there is a small projection I am not accounting for that is being applied by the LGSVL. If anybody has any ideas, I'd be really grateful.

EricBoiseLGSVL, 2021-05-18:
@kamkolak would camera settings be causing this offset?

kamkolak-lge, 2021-05-19:
@jacoblambert 
The issue you're experiencing (ignoring distortion for now) is most likely related to the fact that `camera_world_tf` matrix is not actually `camera -> world`, but instead `camera -> EGO` (all sensors will return their position relative to parent, which in your case means vehicle). Due to active suspension, the whole vehicle settles slightly and camera is lower. This changes perspective for the image you're working on and everything is slightly lower. To take this into account, you should use EGO transform matrix too, which would change the transformation chain to something like this:
```
board space -> world space -> EGO space -> camera space
```

jacoblambert, 2021-05-20:
@kamkolak-lge That was it! Thanks a lot for taking the time to help.
[image]
Indeed the ego to world TF changes a bit _after_ running the simulation in the API. I was actually getting that TF too- but immediately after initializing the vehicle in the API. Of course this was still at the origin until I actually passes the run() command to the simulator.

Sadly I still can't obtain 3D points in distorted images. While waiting for your reply I was trying out Gazebo instead, and run into the same problem in both simulator: the scale and position is off.
LGSVL
[image]
Gazebo
[image]

I understand when using for example cv2's [Optimal Camera Matrix](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga7a6c4e032c97f03ba747966e6ad862b1) and [Rectify Map](https://docs.opencv.org/3.4/da/d54/group__imgproc__transform.html#ga7dfb72c9cf9780a347fbe3d1c47e5d5a) which crops+scales the image so to have a full image, that my 3D->2D projection doesn't apply anymore. So I'm getting something similar is going on here, but feel that probably shouldn't be the case?

In any case, one option I have is to obtain undistorted images and distort them manually so that I know exactly what's going on with the scale, which seems to work well (somehow there's some error around the edges, maybe just rounding...)
[image]

Let me know if you have any comments or ideas about this @kamkolak-lge, if not the issue can be closed. Cheers.

jacoblambert, 2021-05-25:
Thank you very much for your explanation and references.

closed 2021-05-25 by jacoblambert