lgsvl/simulator #1446: Exception: Simulator is not linked. Enter play mode to re-link. CloudAPI+<EnsureConnectSuccess>d__10.MoveNext () (at Assets/Scripts/Web/ConnectionManager.cs:443)

iamshane, 2021-05-18:
After I git clone the simulator, first time to run the simulator , the Unity shows the error Exception: Simulator is not linked. Enter play mode to re-link.
CloudAPI+<EnsureConnectSuccess>d__10.MoveNext () (at Assets/Scripts/Web/ConnectionManager.cs:443). 
I followed all steps the instruction shown, Can you help me why it is like this?
The internet is working.
Please check the picture, thank you.
[image]

EricBoiseLGSVL, 2021-05-18:
Please show some screen shots and player.log to debug the issue. Thanks

iamshane, 2021-05-18:
[image] Here is the error pic in Unity.
And I also check the directory C:\Users\Administrator\AppData\LocalLow\LGElectronics\SVLSimulator
the pic is here:
[image]
unfortunately,  I guess due to cannot run, the player log is not generated.
thank you. @EricBoiseLGSVL

EricBoiseLGSVL, 2021-05-19:
Cool, thanks for the extra info and picture, this helps. Couple of things need addressed.
 - You can't start the simulation from the scene in editor.  You need to fully create the map with needed components, build a bundle and run from webUI.  You start in editor in the LoaderScene.scene in Assets/Scenes.  We have Development Settings in Simulator menu to run a scene in editor but it has many limitations and issues.  Only local prefabs can be loaded and you need to paste old json manually.  It is still wip so I don't recommend it but you can try.
 - I can see that the scene is Untitled, which means it's not saved or is not following our docs on how to make a map.
 - Player log for editor is not in Simulator Appdata but in Unity's Editor Appdata folder.  See [here](https://docs.unity3d.com/Manual/LogFiles.html).

So, if in the editor, start with the loaderscene, link to cloud and then run a simulation.  If you are building a map, refer to the docs on how to do this and upload to cloud.  Hope that helps.

iamshane, 2021-05-19:
I have checked all, thank you for your reply firstly, and I use the .exe running, and after it, all of them works, so thanks again.

closed 2021-05-19 by EricBoiseLGSVL