lgsvl/simulator #1448: Setting an Ego in Offline Mode Simulation

NimrodGoldberg1, 2021-05-19:
When running simulation in offline mode and trying to add agent (sim.add_agent) I receive the following error:
Exception: Could not find asset with Id 47ed7696-82fb-4524-a0c7-4188d6f466a6 (An error occurred while sending the request)

Is there a different function for defining an ego in offline mode?

Thanks!!

EricBoiseLGSVL, 2021-05-20:
Be sure to run the sim online first so the assets are cached.  Let me know if this solves the issue because it could be a bug.

NimrodGoldberg1, 2021-05-21:
I was running first sim online then close connectivity and tried running sim offline. Still got the same issue on add_agent:

Exception: Could not find asset with Id 47ed7696-82fb-4524-a0c7-4188d6f466a6 (An error occurred while sending the request)

Thanks

EricBoiseLGSVL, 2021-05-21:
We will look into this, it could be a bug with offline mode.