lgsvl/simulator #1450: Why did lgsvl use tcp communication?

phs4530, 2021-05-21:
Why did lgsvl use tcp communication?

I know there was udp communication first in Apollo.
When I tried to use lgsvl, it was using tcp communication.
I think udp communication is better when considering the speed of camera images and data. Is there a reason for using tcp communication?
Thank you.

EricBoiseLGSVL, 2021-05-21:
Are you asking about bridge?  Apollo uses Cyber.  What exactly is the issue that you are having?

phs4530, 2021-05-24:
@EricBoiseLGSVL Thanks for reply.
There is no big problem for me. I'm just curious.
My question is why I used tcp to communicate with Apollo.
Reasons for using tcp among udp and tcp.

EricBoiseLGSVL, 2021-05-24:
We assume that you do not want missed data.  AD stacks react poorly when data is lost.  We do support custom bridges so you can create one that uses UDP.  We also have custom sensors you can create that can use UDP if you want.  We have made some UDP sensors for external application integrations.

@hadiTab any specific reason other than packet loss for TCP?

closed 2023-03-10 by phs4530