lgsvl/simulator #1451: Exception: Could not find asset with Id fbeb6e0f-6020-4996-87a5-d56f907ba1ed (BadGateway)

ouissamam, 2021-05-21:
Hello, I am adding EGOs (with ID: Id fbeb6e0f-6020-4996-87a5-d56f907ba1ed) by using the add_agent() method. This is working fine for several episodes, but at certain times I run into the exception: Could not find asset with Id fbeb6e0f-6020-4996-87a5-d56f907ba1ed (BadGateway). It is very strange that it is working fine for several episodes and in the middle of the training it crashes at specific times due to this exception. 
I am also running the same python script with an old version of the LGSVL Simulator and this error does not occur there. So I am assuming that the problem lies on the new version of the simulator.
Sometimes there are the following variations: Exception: Could not find asset with Id fbeb6e0f-6020-4996-87a5-d56f907ba1ed (BadGateway), Exception: Could not find asset with Id fbeb6e0f-6020-4996-87a5-d56f907ba1ed (A task was canceled.), Exception: Could not find asset with Id fbeb6e0f-6020-4996-87a5-d56f907ba1ed (Unauthorized)
Do you know what the problem could be? I am very thankful for any advise.

EricBoiseLGSVL, 2021-05-21:
This sounds like you have something blocking the connection to the WebUI.  We also have an update releasing very soon that might make the connection more stable

ouissamam, 2021-05-25:
Thank you very much for the response. I will check whether I can find something that is blocking the connection to the WebUI. It just seemed curious to me that the same code works fine on an older version of the Simulator and also that on the newest version this happens at specific times. For instance, on the weekend this doesn't happen as frequently as in the middle of the week. I also tried to run the Simulator offline, but this didn't help either. 
I hope that the update makes the connection more stable because I am currently writing my master's thesis and this Exception always crashes the training.

EricBoiseLGSVL, 2021-05-25:
No problem and sorry about the issue you are having.  I'll ping our back-end team on this.  2021.1.1 is released so please try this as well.

ouissamam, 2021-05-26:
Thank you for trying to solve it and pinging your back-end team. Just so you know, I just tried 2021.1.1 but I am having the same issue

marcusvinicius178, 2021-07-29:
Hi @ouissamam  the same is happening to me. And I just say that I solved logging out the simulator account, closing the simulator, cleaning the cache, afterwards redo the simulation (define bridge, Python Api code, etc) and then the simulator will work again...will dowload the scenarios again...
I know it is not a technical solution, but worked for me and I am also in a hurry with my Master Thesis as you are...so I will not try the newer versions hahaha otherwise we can risk don't finish...wish you good lucky.
SVL is amazing team, but I understand that it is pretty hard to manage a lot of bridges, releases,etc...

EricBoiseLGSVL, 2021-07-29:
Yes, we have had some issues finding ways to support the ever growing assets and versions.  We hope to have most problems resolved soon and thanks for posting issues/fixes for others.