lgsvl/simulator #1452: Problems about VR in LGSVL

wobushigaoying, 2021-05-24:
Hellow,

I am using Unity3D version 2019.4.18f1, and LGSVL 2021.01. 

LGSVL is developed based on Unity3D where Unity3D provides the original support for the Oculus VR helmet.

I want to deploy the immersive driving experience, and is it feasible to employ VR helmets in LGSVL.  I mean does the VR helmet work correctly and effectively in LGSVL?

Many thanks.

EricBoiseLGSVL, 2021-05-24:
Yes, but with some work.  I recommend adding VR rig as a sensor plugin and enable VR in the Unity settings of the project.  Let us know if you have issues.

JimWest, 2021-09-24:
I tried to add a VR Rig (Steam VR), it's working but it not rendering anything. I first have to change the Shader of every object (just click on the same shader), than it starts to appear. I don't what the reason is, if I drag a object in the scene when the level has loaded, everything is fine. At this point I don't know if its a unity bug or a lgsvl bug:
[image]

EricBoiseLGSVL, 2021-09-24:
Does this work with the default Unity VR rig?  Steam and Oculus SDK can cause problems in a project. Do you have VR enabled in the project settings?  This might be an HDRP issue.  See [here](https://forum.unity.com/threads/steamvr-with-hdrp.662389/).
You can try a few things:
- be sure to only have one rendering window active.  Unity can be buggy with Scene and Game active at the same time.
- be sure the steam vr cameras have all the required HDRP scripts.  Grabbing the default steam vr rig probably is missing the HDCamera class.  Again this is why it is problematic to use those sdk packages.  They are always incomplete and out of date.
- Last thing would be to look at rendering done in the camera.

Hope this helps.  Please post what other issues/solutions you find