lgsvl/simulator #1454: Unable to set directions in RVIZ

AKSHATA-MK, 2021-05-26:


EricBoiseLGSVL, 2021-05-26:
What errors do you have in Rviz?  Can you screen shot your UI so our devs can see?

AKSHATA-MK, 2021-05-27:
> What errors do you have in Rviz? Can you screen shot your UI so our devs can see?

Thank you for the reply.
I'm sharing the screenshots.
[image]
[image]
[image]

I think map,localisation,sensing and detection are loading well
Let me know what has to be done

EricBoiseLGSVL, 2021-05-27:
@AKSHATA-MK thanks for the images, that helps.  @hadiTab or @lemketron do we support pure-pursuit or any idea what could be wrong?

AKSHATA-MK, 2021-06-03:
Hi 
Now I'm able to set the directions for the vehicle in Rviz
[image]

But the vehicle is stuck 
vehicle is not moving even after the motion planning is launched
I'm sharing some error messages that I'm getting once I'm loading the files, not sure what is effecting the vehicle motion
Once I launch the map.launch file. getting the error as follows, but still then then map is loading , Point cloud is appearing, I'm able to set the directions so not sure if this error is affecting.
[image]

And after I load the localization the timing is not appearing on the right.
After I load the detection.launch the following error is coming.
[image]

Not sure what  is affecting the vehicle movement. It is stuck.
Please Help
Thank you

AKSHATA-MK, 2021-06-03:
I just solved the issue
Thank you

lemketron, 2021-06-03:
Looks like it was missing the ".launch" file?

closed 2021-06-03 by EricBoiseLGSVL