lgsvl/simulator #1455: How to change start point of Apollo in Python API

1Jyfc, 2021-05-26:
Hello!
I'd like to make the ego start at a different start point, and I use the code in this [guide](https://www.svlsimulator.com/docs/python-api/python-api/) to test this. The code use spawns[0].transform as the start point, and it works well with Apollo; but when I change it to spawns[1], the car cannot move. The planning.INFO continuously print that:
```
E0526 20:12:04.671453  8027 reference_line_provider.cc:538] Failed to extract segments from routing
E0526 20:12:04.671459  8027 reference_line_provider.cc:580] Failed to create reference line from routing
```
Is there something wrong in the code? Hope for help. And btw I'd like to know how I can specify the start point and destination by myself without using the spawns, and make sure that the point I specify is correct on line.

EricBoiseLGSVL, 2021-05-26:
Here is a previous issue that will help #1393.  Not sure why apollo wouldn't work with the other spawn point.  What map are you using?  It seems apollo can't localize at the spawn[1] and or the annotation is missing from that spawn. @lemketron ?

lemketron, 2021-05-26:
> It seems apollo can't localize at the spawn[1]

@1Jyfc Are you moving the ego vehicle after initial spawn and with planning already enabled? Apollo 6.0+ does not like that. You'll want to position the ego first and then enable the Apollo modules to ensure that the car is localized properly and more importantly that planning does not lose track of the car as it cannot handle the car teleporting to a new location.

Also what map are you using? As @EricBoiseLGSVL suggested, there may be an issue with annotation. If I remember correctly, if the car does not spawn in the center of a valid lane there may be issues as well.

1Jyfc, 2021-05-30:
Thanks for reply!

I use Borregas Avenue as the map, and its spawns[1] cannot work. And I'll try #1393 recent days. 

And I think I may position the ego before enabling modules, as my code of position like this:
```
ego = sim.add_agent(lgsvl.wise.DefaultAssets.ego_lincoln2017mkz_apollo5_modular, lgsvl.AgentType.EGO, state)
ego.connect_bridge(BRIDGE_HOST, BRIDGE_PORT)
dv = lgsvl.dreamview.Connection(sim, ego, BRIDGE_HOST)
```
, and then enable modules:
```
dv.set_hd_map('Borregas Ave')
dv.set_vehicle('Lincoln2017MKZ')
modules = [
    'Localization',
    'Transform',
    'Routing',
    'Prediction',
    'Planning',
    'Control'
]
destination = spawns[1].destinations[0]
dv.setup_apollo(destination.position.x, destination.position.z, modules)
```
. It works well on spawns[0], but wrong on spawns[1].

EricBoiseLGSVL, 2021-06-02:
Odd, the spawn[1] position is right on the annotation in borregas.  We don't use spawn[1] often but it should work fine like spawn[0].  Let us know what you find when testing.  Can you set a different spawn and get the same results?  Are there errors in DV?

1Jyfc, 2021-06-02:
Yes I'm trying as you said. But I'm new with Unity Editor, could you give me some help about this? I tried to add spawn into the map, but I cannot find the original spawns of BorregasAve in the map, so I don't know if I can successfully add a spawn into it in my way. I open the LoaderScene and use `Import HD Map` to load the map `/apollo/modules/map/data/borregas_ave/base_map.bin`, and in `Annotate HD Map` mode I cannot find the spawns, while they are not in Hierarchy either. Could you tell me whether my way is wrong? Because I don't find docs from [here](https://www.svlsimulator.com/docs/) about this problem. Hope for help.

EricBoiseLGSVL, 2021-06-02:
@1Jyfc Since Borregas Ave is not an open source map, you cannot edit the spawn positions.  HDmaps do not have spawn info data, they are a simulator Unity component.  If you want to change the spawn position, you will need to run the scene in editor, find the position you want, and then spawn with the api. You don't need to load the hdmap in the loader scene. Use the loader scene to start a simulation with borregas.

1Jyfc, 2021-06-16:
Thanks for help and sorry for replying so slow! Recent days I learned to use Unity to generate the scene and tried to use BorregasAve to generate a simple scene to test if apollo ego can start from different spawn positions. And it still cannot work. I tried to make ego start from a spawn which location is [5.84999990463257, -1.75999999046326, 20.2900009155273]:

[image]

The ego can run till the crossroad, and it stops with these log:
```
E0616 20:47:39.941265  8127 st_bounds_decider.cc:316] No valid st-boundary exists.
E0616 20:47:39.941516  8127 stage.cc:108] Failed to run tasks[ST_BOUNDS_DECIDER], Error message: No valid st-boundary exists.
E0616 20:47:39.941612  8127 speed_profile_generator.cc:38] Fallback using piecewise jerk speed!
W0616 20:47:39.941680  8127 speed_profile_generator.cc:41] init_v = 0, init_a = 0
W0616 20:47:39.941694  8127 speed_profile_generator.cc:47] Already stopped! Nothing to do in GenerateFallbackSpeed()
```

But if I set such position and rotation in BorregasAve scene that you offer, it can run correctly. Still spawns[1] is not OK, but for me it's enough to set the position by editing the spawn with code. And I still have two questions:

1. Do you use windows to run SVL and Linux to run Apollo on two devices? Or run SVL and Apollo on the same Linux device like me? I think the problem that spawn[1] cannot work may be caused by hardware problems.
2. Have you ever considered to provide your BorregasAve on github? This request may be excessive, but I think those researchers who use SVL to solve problems that are not related to scene construction will need it very much, like me. Hope that you could consider of it, and thanks for the help so far.

EricBoiseLGSVL, 2021-06-21:
@hadiTab are we adding any spawn data to the map.bin file when we export?