lgsvl/simulator #1456: Opening test results in webUI causes an error and logout

krtt, 2021-05-26:
In the webUI, when clicking on the 'Test Results' in the left menu, sometimes shortly shows an error on the page (where usually the results are displayed) and then quickly logs you out. Difficult to reproduce right after it occurs, but has happened approx 5 times in total (3 weeks of usage). The exact error message is also difficult to capture because logout happens fast.

EricBoiseLGSVL, 2021-05-26:
@zelenkovsky any idea how to capture this error?

thisisdavidfrank, 2021-05-26:
@krtt, based on your description, it looks like either the session timed out (since there isn't any error response from the server that can automatically log you out by force), or there is something else that makes it appear that you have been logged out. When this happens to you the next time, can you simply hit the browser back button to go back to the previous page and verify that you are still logged in, please?

krtt, 2021-05-31:
Today I downloaded the new version (2021.1.1) and next time it happens I will recheck with the browser back button.

krtt, 2021-06-27:
It has not occurred in the new version so I think you can close this issue.

closed 2021-06-28 by EricBoiseLGSVL