lgsvl/simulator #1457: Creating and running API-only simulation results in logout

krtt, 2021-05-26:
In the webUI, after creating an API-only simulation following [this guide here](https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation), running this sim will briefly show the status as "API ready!" but then goes to logout and offline. Resulting status snapshot:
[image]

EricBoiseLGSVL, 2021-05-26:
Do you have the player.log so we can see the error?  You can also see an error in webui with f12

krtt, 2021-05-31:
No player.log any more unfortunately. But I have a new problem. Will post as a separate issue..

krtt, 2021-07-15:
I have used the API Only sim for a while now and the problem has not appeared any more. It's ok to close this issue.

closed 2021-07-15 by krtt