lgsvl/simulator #1459: What is the interface to get sensor raw data

DanielTian90, 2021-05-27:
Hi LGSVL,
I am prepared to use lgsvl as simlation scenario to inject data to my adas ECU. But I didn't find the interface of data from sensor(eg. radar/camera/lidar) of lgsvl simulator. Could you please to tell me regarding detailed definition for these data? Documantation or source code.
Many thanks.

EricBoiseLGSVL, 2021-05-27:
ROS, ROS2, CYBER bridge.  Documentation is [here](https://www.svlsimulator.com/docs/system-under-test/simulator-messages/).

DanielTian90, 2021-05-28:
I want to know detailed definition for each topic, eg. for lgsvl_msgs/DetectedRadarObject, I want to know definition for data format.

EricBoiseLGSVL, 2021-05-28:
For that you can look at the source code.  These are defined in the bridge code.
[image]

DanielTian90, 2021-05-28:
OK.
Regarding attributes in lgsvl_msgs/DetectedRadarObject, these data is for detection point level before classification? or after classification(eg. attributes for car/pedestria/bicycle)?

EricBoiseLGSVL, 2021-05-28:
No classification is done on radar objects.  This is just the raw data from detected objects as you can see from the data class.  positions, velocities, angles and size, no classification.

closed 2021-05-31 by DanielTian90