lgsvl/simulator #1460: Failed to build archive, exception follows:

duongtran11, 2021-05-27:
I got errors when building a custom ego vehicle
[image]

I got the same errors when building a JaguarXE2015 downloaded from github
[image]

What could be the problem?

EricBoiseLGSVL, 2021-05-27:
Do you have both Windows build support installed?
[image]

If yes, then it seems an issue with FBX2glTF converter plugin on your machine.  You might be missing a needed library for it to work.  Can you post the full stack trace here please?  What version of Ubuntu are you on?

Last thing to check.  Did you name the prefab/folder the same?  Is anything blocking unity access to these AssetBundle folders?  What does the asset bundle look like when it fails (screen shot the folder holding the asset bundle)

EricBoiseLGSVL, 2021-05-27:
Any idea if Linux needs something for FBX converter? @kamkolak

duongtran11, 2021-05-28:
I did name the prefab and the folder that contains it the same. I have build ego vehicles multiple times before, on previous version (LGSVL running on Unity Editor 2019.3.3f1), never have this error.
I have windows support installed
[image]

and this is the folder stack
[image]
I'm on Ubuntu 18.04
Should it works fine on Windows?

EricBoiseLGSVL, 2021-05-28:
thanks for the images.  It seems you have a few extra modules.  On linux you only need windows support, not UWP and I'm not sure what the center one is, webGL? As for the folder, can you show the output folder with the bundles.  When it fails it lists the files and we can tell sometimes from the leftover files.  I want to see if it is saving the gltf but naming it something wrong.

duongtran11, 2021-05-28:
I think it's not UWP, it's Linux build support (IL2CPP), the center one is WebGL  build support. 
Here is the output folder (in the process of building)
[image]
meanwhile in the Unity Editor window
[image]

and here is after the process of building has finished
[image]
any other files that created during the process of building disappeared

and the Unity Editor window
[image]
[image]

EricBoiseLGSVL, 2021-05-28:
Thanks for the images. Sorry but we need to dig deeper.

It should be Linux Build support Mono (the version you downloaded from Unity) and Windows Mono support only.  Uninstall the other modules

Please post the editor player.log.  I cannot see the full stack trace from an image of the editor console window.

Last image we need is the bundle because it looks like it builds fine.  Are you able to upload it to webUI and run a sim?

Look in the assetbundle like so and post also
[image]

The arrow points to the folder that should have a .glb like so
[image]

What is in your folder?

duongtran11, 2021-05-31:
Here are modules that I've installed, didn't have Linux build support (mono)
[image]

Here is the player.log
[Player.log](https://github.com/lgsvl/simulator/files/6567345/Player.log)

After the build completed, there is only one file in the AssetBundles/Vehicles, and that file's size is 0kb, so I couldn't upload it to webUI
[image]

As one of the error that I've captured above said, the .glb file was missing.
[image]

kamkolak-lge, 2021-06-01:
@EricBoiseLGSVL Autodesk FBX package requires some shared objects to be present, but I think they should be included with default Ubuntu 18.04 installation. Here's the list:
```
UnityFbxSdkNative.so
        linux-vdso.so.1 (0x00007fff0c94c000)
        libxml2.so.2 => /lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f2d096b1000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2d09695000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2d094b4000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2d09365000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2d0934a000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2d09158000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2d09150000)
        libicuuc.so.66 => /lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007f2d08f6a000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f2d08f41000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2d0a142000)
        libicudata.so.66 => /lib/x86_64-linux-gnu/libicudata.so.66 (0x00007f2d07480000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2d0745d000)
```

@duongtran11 the player log you attached does not contain any of the exceptions or errors you listed. Are you sure it was the same session?

duongtran11, 2021-06-02:
@kamkolak-lge thanks for the reply!
Here are player log and editor log, right after I finished my build
[Player.log](https://github.com/lgsvl/simulator/files/6581762/Player.log)
[Editor.log](https://github.com/lgsvl/simulator/files/6581763/Editor.log)

Are you saying I need to install all those libraries?

kamkolak-lge, 2021-06-02:
@duongtran11 The most likely reason for this issue is FBX converter failing to launch. Please check if you have all the dependencies available on your system, and add any that are missing.

duongtran11, 2021-07-03:
@kamkolak-lge Or maybe because I don't have Simulator.Vehicles.asmdef file in External/Vehicles?
I just clone new simulator and successfully build the vehicle. I notice that the new repo that I just clone have Simulator.Vehicles.asmdef file in External/Vehicles
Maybe that the different??
But I still need to build vehicles with 2021.1.1 version

kamkolak-lge, 2021-07-06:
@duongtran11 Assembly definition file should not be a problem, unless you're including some custom code in your vehicle bundle.

Since you were able to build on newer version, you have all required dependencies on your system. Looking at all the new information, I think the most likely cause for the problem is race condition. If it's possible and won't cause compilation issues, please try cherry-picking commit [`80c0919f195bc1e94c37abbe4242e2cf7377ca96`](https://github.com/lgsvl/simulator/commit/80c0919f195bc1e94c37abbe4242e2cf7377ca96) onto 2021.1.1 branch. Alternatively, you could just modify `Assets/Scripts/Editor/Build.cs` file on 2021.1.1 version:

After the [line 262](https://github.com/lgsvl/simulator/blob/e844604459f04c0f0dbf007f264eaf9618d32f88/Assets/Scripts/Editor/Build.cs#L262):
```
p.Start();
```
try adding:
```
p.WaitForExit();
```

Please let me know if this solved your issue.

duongtran11, 2021-07-08:
I tried what you recommended, still get the same error
[image]
There's a warning about a .fbx file not found
This is my folder tree
[image]
[image]

duongtran11, 2021-07-08:
The reason why I have to build vehicles in 2021.1.1 is because my environment is in com.svlsimulator.2 version and the vehicles built by newer simulator versions are in com.svlsimultor.5 version. So it got a version mismatch when I ran simulation.
I can build environment with newer simulator version but then it would mismatch with sensors version (com.svlsimulator.3) 
I can build most of the sensors to the newest version but Autoware Car control sensor doesn't have asset files to build

EricBoiseLGSVL, 2021-07-08:
So the issue is with the glb library on your machine so you have to build assets with the latest version, right?
The latest version fixes the issue, correct?
I don't see how you can stay on 2021.1.1 when you can't build assets locally with your rig.
Let's look into why AutowareAIControlSensor repo is not public.  @lokesku @hadiTab @eagafonovlge

duongtran11, 2021-07-09:
@EricBoiseLGSVL well, I made an environment on 2021.1.1 first, then proceeded to build a vehicle, then got the error. I tried to switch to newer version of simulator but got other problems with version mismatch.
Are you planning to upgrade sensors to the newest version?

EricBoiseLGSVL, 2021-07-09:
Sorry, about these issues with the release.
All assets are updated for latest release.  Sorry about the confusion.  Please delete all asset bundles in the AssetBundles folder in the root, delete the AppData folder here, ```C:\Users\YOU\AppData\LocalLow\LGElectronics\SVLSimulator``` and finally unlink and relink the cluster.  Build all assets to the new release 2021.2.  Be sure to change the config.yml url to wise.svlsimulator.  This should resolve your issues.

duongtran11, 2021-07-10:
I'm getting the same error as #1566

EricBoiseLGSVL, 2021-07-12:
The issue is being fixed now, it only effects users that are building binaries locally.  Sorry for the confusion.

duongtran11, 2021-07-16:
I've successfully build the vehicle, thank you!

closed 2021-07-16 by duongtran11

CompFreakAlpha, 2021-07-25:
> The issue is being fixed now, it only effects users that are building binaries locally. Sorry for the confusion.

@EricBoiseLGSVL What was the "fix" for this? I'm seeing the exact same problem trying to build a vehicle (Jaguar, unmodified) on _Windows_ with 2021.2:

> FileNotFoundException: Could not find file "C:\Users\CompF\Desktop\Unity Projects\simulator\AssetBundles\Vehicles\dc850ed4-a8d0-4e1e-bffb-cb6da5413e82_vehicle_Jaguar2015XE.glb"
> System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) (at <eae584ce26bc40229c1b1aa476bfa589>:0)
> System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <eae584ce26bc40229c1b1aa476bfa589>:0)
> (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare)
> System.IO.File.Open (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <eae584ce26bc40229c1b1aa476bfa589>:0)
> ICSharpCode.SharpZipLib.Zip.StaticDiskDataSource.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
> ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
> ICSharpCode.SharpZipLib.Zip.ZipFile.AddEntry (ICSharpCode.SharpZipLib.Zip.ZipFile workFile, ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate update) (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
> ICSharpCode.SharpZipLib.Zip.ZipFile.RunUpdates () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
> ICSharpCode.SharpZipLib.Zip.ZipFile.CommitUpdate () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
> Simulator.Editor.Build+BundleData.RunBuild (System.String outputFolder) (at Assets/Scripts/Editor/Build.cs:783)
> UnityEngine.Debug:LogException(Exception)
> Simulator.Editor.BundleData:RunBuild(String) (at Assets/Scripts/Editor/Build.cs:789)
> Simulator.Editor.Build:BuildBundles(String, Action) (at Assets/Scripts/Editor/Build.cs:1132)
> Simulator.Editor.<>c__DisplayClass18_0:<OnGUI>b__1() (at Assets/Scripts/Editor/Build.cs:950)
> UnityEditor.EditorApplication:Internal_CallDelayFunctions()

EricBoiseLGSVL, 2021-07-26:
We are looking into this.  Looks like our fix couldn't go to linux so it was rolled back.  We will follow up.