lgsvl/simulator #1461: Problem about map annotation

wobushigaoying, 2021-05-31:
Hello team,

I am using map annotation tools following the instruction.

I cloned the CubeTown from the github for the test. I opened the CubeTown.scene in Unity editor.

As the below figure indicates, I clicked "View All" in "View mode", and "Lane/Line" in "Create Modes". TARGET_WAYPOINT is then showed up in the hierarchy tab, but no yellow indicator in my scene. If I click TARGET_WAYPOINT in the hierarchy tab, it disappeared.

 I checked the "Road" and "Mesh collider" had been attached yet. What's wrong with this??

[image]

[image]

wobushigaoying, 2021-05-31:
Hello, I have solved this problem. I enabled the Gizmos button in the Unity editor, and then the WAYPOINT appeared. I have successfully created my own MapLanes. I am going to create MapLanes in our own Maps.  :)

[image]

EricBoiseLGSVL, 2021-06-02:
Nice, sorry that this was not toggled by default.  We will update the docs as soon as we can. Thanks for letting us know.

closed 2021-07-16 by EricBoiseLGSVL