lgsvl/simulator #1463: Cannot correctly stop the Python Test Case Simulation

krtt, 2021-05-31:
I followed the manual [here](https://www.svlsimulator.com/docs/running-simulations/running-simulator/#pythontestcasesimulation) to test the "Cut-in Scenario" and when I get to the end of point 7 (click Run simulation), then the sim starts, but after ca 10-20 seconds it stops in the sim window (and displays the 'Open browser' and 'Visual scenarios' buttons), but at the same time in the browser web UI, it shows still 'running' and the stop button there does not work. 
[image]

The error on the sim name (in the web UI) says 'Error: Cluster is currently running another simulation'.
If I close everything and run again, same result.

At this point I did not even get to the point nr 9 in the guide (to go to Apollo Dreamview to configure it).

EricBoiseLGSVL, 2021-06-02:
Are you using 2021.1.1 release?

krtt, 2021-06-02:
Yes, I am using 2021.1.1, prebuilt (downloaded from svlsimulator.com page)

EricBoiseLGSVL, 2021-06-02:
First, do you have the bridge started before you run the simulation? Second, please post your player.log to see any errors that you have.  Lastly, what OS are you using?
@eagafonovlge @thisisdavidfrank

krtt, 2021-06-07:
Yes, the bridge was started before running the sim.
Player.log attached.
[Player.log](https://github.com/lgsvl/simulator/files/6607420/Player.log)
Ubuntu 18.04.

EricBoiseLGSVL, 2021-06-07:
```
SocketException: Connection refused
  at System.Net.Sockets.SocketAsyncResult.CheckIfThrowDelayedException () [0x00014] in <ef151b6abb5d474cb2c1cb8906a8b5a4>:0 
  at System.Net.Sockets.Socket.EndConnect (System.IAsyncResult asyncResult) [0x0002c] in <ef151b6abb5d474cb2c1cb8906a8b5a4>:0 
  at Simulator.Bridge.Cyber.CyberBridgeInstance.OnConnectionComplete (System.IAsyncResult ar) [0x00000] in <07e3d881d39e45158bcdba3f8cecc23d>:0 
```

This is the error.  Any idea why the connection is refused on your machine?  Are you using Apollo 5.0?

krtt, 2021-06-08:
No idea why connection refused. In some other cases it runs fine. Yes, Apollo 5.0.

EricBoiseLGSVL, 2021-06-08:
does this happen after running a sim?  We have seen apollo need to be completely closed before another run.  Right @hadiTab @lemketron ?

krtt, 2021-06-16:
I have not been able to reproduce this situation any more. It´s now stuck in a new situation. I will put a separate issue about this.

closed 2021-06-21 by EricBoiseLGSVL