lgsvl/simulator #1464: Problem about connecting two lanes

wobushigaoying, 2021-06-01:
Hello team... 

I am using CubeTown as a case study for map annotation.

The two lanes are surely connected in the scene as shown in Figures 1 and 2.  Could you please help me connecting the lanes as shown in Figure 3. I tried to add a junction between these two lanes, but it seemed not to work ( I added two waypoints on the two waypoints of these two lanes, however, the "Connect" button did not work. ). 

Many thanks!!!

[image]

[image]

[image]

EricBoiseLGSVL, 2021-06-02:
connect button is for two temp waypoints to make one lane.  Not to connect two lanes together.  If you want to extend the lane, select the maplane object and enable display handles so you can move the point to where you want.  If you want two separate lanes, then move the first index point to overlap the last index point of the prev lane.  Like so:
[image]

wobushigaoying, 2021-06-02:
OK!!!  Thanks for your kindly reminder. ~  :)

closed 2021-07-16 by EricBoiseLGSVL