lgsvl/simulator #1468: Does the SVL support simulation of robot arms, e.g. UR10?

manone888, 2021-06-02:
Hi guys, your webpage https://www.svlsimulator.com/use-cases/#robotics mentions Robotics simulation. I'd like to know if I could control a robot arm, e.g. UR10 via ROS in the SVL simulator. This will require sending commands from ROS nodes to the simulator and sending joints' positions from the simulator to ROS nodes.

Thank you. I look forward to your reply.

EricBoiseLGSVL, 2021-06-03:
Yes, we are working on the next release that supports articulated bodies.

manone888, 2021-06-03:
Hi Eric, thank you for your reply. 

It sounds great! Do you have an estimated time for the next release?

EricBoiseLGSVL, 2021-06-03:
Estimated end of this month

manone888, 2021-06-03:
Wow, that would be so great! I really look forward to it.

Thanks a lot.

ShawnSchaerer, 2021-08-06:
Any updates on when this will be supported?

EricBoiseLGSVL, 2021-08-06:
@ShawnSchaerer testing the release all this week :)

ShawnSchaerer, 2021-08-06:
> @ShawnSchaerer testing the release all this week :)

Great, thanks!