lgsvl/simulator #1469: When creating simulation scenarios of Python API, could ros2 bridge be used?

DZiYou, 2021-06-03:
When I run a python API simulation scenario，want to send and receive data through ros2 bridge,but after run lgsvl_bridge
only 
[lgsvl-bridge]: Listening on port 9090
could not 
[lgsvl-bridge]: Client [127.0.0.1:57392] connected.
Is there something wrong with my settings? How to check to run ros2 bridge？

DZiYou, 2021-06-03:
the python code ,and had set localhost:9090 in autopilot

`from environs import Env
import lgsvl

env = Env()

LGSVL__SIMULATOR_HOST = env.str("LGSVL__SIMULATOR_HOST", "127.0.0.1")
LGSVL__SIMULATOR_PORT = env.int("LGSVL__SIMULATOR_PORT", 8181)
LGSVL__AUTOPILOT_0_HOST = env.str("LGSVL__AUTOPILOT_0_HOST", "127.0.0.1")
LGSVL__AUTOPILOT_0_PORT = env.int("LGSVL__AUTOPILOT_0_PORT", 9090)

sim = lgsvl.Simulator(LGSVL__SIMULATOR_HOST, LGSVL__SIMULATOR_PORT)
sim.load(env.str("LGSVL__MAP"))

egoState = lgsvl.AgentState()
ego = sim.add_agent(env.str("LGSVL__VEHICLE_0"), lgsvl.AgentType.EGO, egoState)

npc = sim.add_agent("Sedan", lgsvl.AgentType.NPC)
waypoints = [
  lgsvl.DriveWaypoint(lgsvl.Vector(0,0,0), 0, lgsvl.Vector(0, 0, 0), 0, False, 0),
]
npc.follow(waypoints, loop=True)

sim.run()`

EricBoiseLGSVL, 2021-06-03:
I'm not clear on your issue.  Are not able to connect to the bridge?
@hadiTab  @lemketron ?

lemketron, 2021-06-03:
Could you share more specifics about the simulation you are running?  Did you create a "API Only" simulation and are connecting to it with a stand-alone python script?  Or are you using a "Python API" simulation and pasting the script into the simulation web site?

Have you looked at the following documentation which may be helpful?
https://www.svlsimulator.com/docs/system-under-test/autoware-auto-instructions/
https://www.svlsimulator.com/docs/running-simulations/running-simulator/

EricBoiseLGSVL, 2021-06-04:
@DZiYou please try the script with API Only simulation, not the PythonAPI mode.  Seems we have a bug and hope to have a fix soon.

DZiYou, 2021-06-04:
> @DZiYou please try the script with API Only simulation, not the PythonAPI mode. Seems we have a bug and hope to have a fix soon.

Thank you for your reply.
When using PythonAPI only mode, how to set ROS2 bridge through Python?Is there a reference document？
I'll change it to  
LGSVL__AUTOPILOT_0_HOST = env.str("LGSVL__AUTOPILOT_0_HOST", "127.0.0.1")
LGSVL__AUTOPILOT_0_PORT = env.int("LGSVL__AUTOPILOT_0_PORT", 9090)

sim = lgsvl.Simulator(LGSVL__AUTOPILOT_0_HOST, LGSVL__AUTOPILOT_0_PORT)

But Python can't connect next

EricBoiseLGSVL, 2021-06-04:
You should use APIOnly simulation.  If this is your ip and port then, yes it is how you set it.  Make sure no other applications are using the port and start bridge before running api.

DZiYou, 2021-06-10:
> You should use APIOnly simulation. If this is your ip and port then, yes it is how you set it. Make sure no other applications are using the port and start bridge before running api.
When I want use APIOnly and build ROS2 bridge,Can any vehicle be used？
there is a ERROR: 
Agent 'ac4e290a-9f48-4748-bb4d-410f8621f676' is missing bridge client

EricBoiseLGSVL, 2021-06-10:
Then that vehicle doesn't have a bridge set.  Set the bridge on the vehicle.

DZiYou, 2021-06-10:
> Then that vehicle doesn't have a bridge set. Set the bridge on the vehicle.

@EricBoiseLGSVL 
 @[image]
This is the configuration of Vehicle Sensor , had set bridge but how select 'acc' sensor use PythonAPI.

EricBoiseLGSVL, 2021-06-10:
This is just the vehicle sensor config.  sensor is not set to python api.  You copy the sensor config id and use that as the vehicle guid in the python api.
[image]