lgsvl/simulator #1471: Problems with 2021.1 built on Ubuntu 20

henrychoi, 2021-06-04:
Hello, thank you for the new release, which I am told (2nd handedly) fixes the bug in ImuSensor (multi-threading) and the steering (the reported angle is always the maximum values).

Just pulled the branch release-2021.1 and built it on the same Ubuntu 20 node where I was running the previous simulator (also built locally), but I get the following errors:
```bash
henry@z97:~/simulator$ build/simulator
Set current directory to /home/henry/simulator
Found path: /home/henry/simulator/build/simulator
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  99
  Current serial number in output stream:  100
```
Another (unrelated problem): what happened to the ImuSensor.cs?

henrychoi, 2021-06-04:
About the ImuSensor and many other Assets that were deleted from the simulator latest release: it looks like they were broken out as separate repos.  Are we supposed to clone what we need into the Assets/External/<type> folders one by one?  Or should they be cloned to Assets/Scripts/<type>?

EricBoiseLGSVL, 2021-06-04:
Clone to External/Sensors/
2021.1.1 has no changes to core, seems like your environment is messed up.  I'd try to build from simulator menu, see if that works.

closed 2021-07-24 by henrychoi