lgsvl/simulator #1473: LGSVL 2021.1.1 can't start on Ubuntu 18.04

ghostsun89, 2021-06-04:
When starting the pre-compiled version(svlsimulator-linux64-2021.1.zip) downloaded from the release section it fails to start. The error infomation is as follows :

Terminal:
```
$ ./simulator          
Set current directory to /home/maxandre/Downloads/lgsvlsimulator-linux64-2020.06
Found path: /home/maxandre/Downloads/lgsvlsimulator-linux64-2020.06/simulator
The process stops and nothing.
```

Player.log:
```
Mono path[0] = '/home/svlsimulator-linux64-2021.1.1/simulator_Data/Managed'
Mono config path = '/home/svlsimulator-linux64-2021.1.1/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'default 42"': 3840x1080 (primary device).
Loading player data from /home/svlsimulator-linux64-2021.1.1/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 3840 x 1080 @ 0 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
Caught fatal signal - signo:11 code:1 errno:0 addr:0x8
Obtained 12 stack frames.
#0  0x007fdbb6caf980 in funlockfile
#1  0x007fda216668d0 in _init
#2  0x007fda2166a146 in _init
#3  0x007fda24d7fb70 in _init
#4  0x007fda24d7ffb2 in _init
#5  0x007fda24d7f6cf in _init
#6  0x007fda24d82435 in vkEnumeratePhysicalDevices
#7  0x007fdbb7b2bb2b in vk::CheckVulkanSupport()
#8  0x007fdbb7fd02b7 in IsRendererSupported(GfxDeviceRenderer, GfxDeviceLevelGL, void*)
#9  0x007fdbb7fcfc6d in PlayerMain(int, char**)
#10 0x007fdbb68cdbf7 in __libc_start_main
#11 0x00000000400569 in _start
```

Specifications :

Ubuntu 18.04
Tesla V100
Nvidia driver version: 440.95.01
CUDA Version: 10.2

lemketron, 2021-06-04:
Can you try running `vkvia` (you may need to first `sudo apt install lunarg-via`) and see what that returns? It should tell you what GPU Vulkan is trying to use.

Even if you have installed libvulkan, if you also have mesa-vulkan-drivers installed then that can cause problems.

If you have "mesa-vulkan-drivers" installed that can cause problems when vulkan tries to use the Nvidia GPU.

You can remove this with `sudo apt remove mesa-vulkan-drivers`.

ghostsun89, 2021-06-07:
@lemketron Thanks for your suggestion. Through the cmd ```vkvia```, I got the wrong info as follows:
```
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
dpkg-query: no packages found matching vulkan-sdk
VIA_ERROR:   Vulkan failed to find a compatible driver.
```
It looks like the uncompatible driver caused the problem. I will try to update the driver later.

closed 2021-06-07 by ghostsun89