lgsvl/simulator #1475: Build Apollo 3.0 - Missing ros_pkg or cyber folder

marcusvinicius178, 2021-06-04:
Hi I am trying to build Apollo 3.0 alongside the SVL simulator. I have already set the Apollo 3.0 custom ROS Apollo bridge inside SVL according to @hadiTab:

[image]

Following this tutorial: https://www.svlsimulator.com/docs/archive/2020.01/apollo-instructions/

Speciffically in step:
`git clone --recurse-submodules https://github.com/lgsvl/apollo.git`

If I clone this repo I am going to have just the cyber folder/ cyber briged, and not the ros_pkgs folder and the rosbridge which is required later for a subsequent step present in his tutorial:

[image]

I have found in @hadiTab repo the ros_pkgs folder: https://github.com/lgsvl/apollo-3.0
This repo is different from the tutorial repo: https://github.com/ApolloAuto/apollo

Then I have cloned both repo inside my Apollo 3.0 created folder:

[image]

However to start the images:

`./docker/scripts/dev_start.sh` 

I must be inside apollo or apollo-3.0-simulator folder. And as the first does not contain the ros bridge and the second does not contain the cyber bridge. I ask myself which one I must run this command, in order to have a correct connection on Dreamview- SVL simulation ?

Trying the first option (start sh from apollo folder) Dreamview does not receive info from the simulator
Trying the second option (start sh from apollo-.0-simulator folder)  I get the error below?

[image]

I guess that should exist just one repo (merged of apollo and apollo3-0-simulator) to be cloned and used.
Could someone assist me how to propely build apollo 3.0 along SVL? 
The tutorial is wrong and need to be updated!

Thanks in advance

marcusvinicius178, 2021-06-04:
The files inside the ros_pkgs folder are not dowloaded automatically for some reason:
https://github.com/lgsvl/apollo-3.0/tree/simulator/ros_pkgs/src

The folders rosbridge_suit and lgsvl_msgs come empty . Then I needed to dowload them manually from repo. Afterwards I have built the workspace again with catkin_make. Now ROS says that I do not have the geometry_msgs pkg installed. However it is not possible install anything inside Apollo container.
Always when I issue some cmd line, such as "sudo apt-get install ros-indigO-..." I get errors...

lemketron, 2021-06-07:
> Could someone assist me how to propely build apollo 3.0 along SVL?
> As a solution could I copy and past the **ros_pkgs** folder from this repo https://github.com/lgsvl/apollo-3.0 to this repo folder https://github.com/ApolloAuto/apollo ? Would this work?
> 
> The tutorial is wrong and need to be updated!

Apollo 3.0 is two years old and we don't really have the time to support it especially with SVL Simulator 2021. We did publish the ROS Apollo bridge for SVL but can't spend much time supporting such an old version of Apollo.

There are some archived docs for using Apollo 3.0 with LGSVL Simulator 2020.06 simulator which may be helpful: https://www.svlsimulator.com/docs/archive/2020.06/apollo-instructions/

I really would recommend focusing on newer versions of Apollo like 6.0 or master, which support the latest Nvidia GPUs and will receive the most testing and support with SVL Simulator 2021.x going forward.

marcusvinicius178, 2021-06-08:
Hi Steve thanks for the tutorial. I have already used it and fails. But not due wrong given steps, but due the correct Branch and tag to be checkout. There are additional steps that the user must do:

`git checkout tags/v3.0.0`

For example enable the existence of the ros_pkgs and afterwards the catkin_make commands finally works... HOWEVER the bridge.sh file keeps missed...and it is not possible to run the Ros-Bridge.

The reason why I needed to downgrade to use Version 3.0 than 6.0 is to the fact that until this moment it was possible to switch the path planners from EM (Road_public) to Lattice just on 3.0 version on conf folder/ planning_config.txt.proto file 
In newer versions when the user change this line of code in order to switch the planner, the planning tab on Dreamview is disabled...

As I said a experienced Apollo user helped me to switch planner on Apollo 3.0, using the sim_mode (internal simulator) just checking out the branch. And then I also would like to test the vehicle behaviour with the LG simulator (not just on Dreamview). 

I confess I still did not try to checkout from master to tags/v 6.0 on Apollo6.0 to see if the planners can therefore be switched with success. If works ok it is good! However this user explained me to change planners in Apollo 6.0 it is not a so easy task, once specific scenarios and tasks will not accept and be configured for the Lattice Planner and therefore it will keep not working (even with the checkout on tag).

Do you have some experience in changing planners on Apollo5.0 or 6.0? Please let me know. I am very interested to achieve that!

closed 2022-03-13 by marcusvinicius178