lgsvl/simulator #1476: LGSVL or SVL simulator can provide Graphics?

marcusvinicius178, 2021-06-06:
Hi @EricBoiseLGSVL , @lemketron I would like to understand if LGSVL or SVL simulators could provide graphics from the car behavior while it is taking command from planning or control module (from Apollo AD stack or Autoware)?

It would helpe me a lot, considering I am comparing 2 planners in Apollo: EM and Lattice. However Apollo it seems that Apollo team did not debug for the Lattice planner and does not generate graphics as you can check in the issue below:
https://github.com/ApolloAuto/apollo/issues/13837

For this reason I need other "source" to get graphics and compare these planners. I am aware tha SVL provides the video_recording called birdview on SVL, while the user set a custom autopilot (from the bridges available: apollo or autoware). 

Would you know to say if exist other options from the simulator side, to plot data while the car is traveling (path, trajectory, odometry, acceleration, brake, Lidar data). Any data is welcome. 

I ask this because from the simulator panel, clicking on the "eye" icon it displays the data coming from the AD stack Topics...

If you could think on a solution for me get this data visually, please let me know. I really need them to complete the thesis...

EricBoiseLGSVL, 2021-06-07:
Sure, you can record the data from the bridge and input into any graphing application using timestamps.  We often do this to test performance.

marcusvinicius178, 2021-06-07:
That is pretty nice @EricBoiseLGSVL how can I get this bridge data? Could you forward me please if possible a video/tutorial example?

EricBoiseLGSVL, 2021-06-07:
You subscribe the the topics and save the data.  [This](https://www.svlsimulator.com/docs/tutorials/lane-following/#collect-data-from-svl-simulator) is the lane follow example that uses ROS.

marcusvinicius178, 2021-06-07:
Nice and for Apollo6.0 that uses just cyber_rt some example  to get the bridge data?

lemketron, 2021-06-07:
> That is pretty nice @EricBoiseLGSVL how can I get this bridge data? Could you forward me please if possible a video/tutorial example?

Check the Cyber_recorder documentation here: https://github.com/ApolloAuto/apollo/blob/master/docs/cyber/CyberRT_Developer_Tools.md

marcusvinicius178, 2021-06-08:
Hi Steve Thanks for the documentation! It is very complete. And I am feeling a similar operational cmds of ROS and Cyber regarding recording topic.Please share me the your last recommendations to close this issue: 

I am considering get this data, convert to csv file and plot using matplotlib on Anaconda:

1- Do you know/suggest another more easier/quicker strategy/software to plot and display data with a nice visualization?

From previous experience using ROSBAG I've generated a CVS file containeing million of lines (in a very short record period) . I believe the freq to get data is set standard to be too high? 

2- Is it possible get less amount of  data/second with cyber RT?  Is the command, below the responsible to control this data flow or another? 

`cyber_record record -i 2`

-i = record segment interval in n seconds
In the example above I would record data from topic just after 2 seconds....I am looking for a reasonable freq to get data and display. 

3- is there a standard frequency to get vehicle data (from odometry, gnss, etc?). Which is?

marcusvinicius178, 2021-07-16:
Hi @lemketron and @EricBoiseLGSVL 

I needed to reopen this issue for more assistant if possible. The link Steve sent me is good, however not enought, it is a little bit old for Apollo 2.5 version where Baidu team still used rosbag, so it was possible to get the rosbag data and convert to csv, txt formats for later assessment and plot.

Now the Baidu just user CyberRT, which make things harder to me, once this .record file is in binary format and is not possible to convert it directly to .csv or .txt or any other readable format. Thus it is not possible to get this data so easily. I have answered this on Apollo git also, and I got some answers regarding that I should use :

1) A Listener.Py : https://github.com/ApolloAuto/apollo/blob/master/cyber/python/cyber_py3/examples/listener.py 
To save the channels data online, however this requires a bunch of code and adapt the already available code to all other channels, which is not so trivial (As a ROS developer I am experienced in face a bunch of errors while adjsuting/reusing a module of code to other sensors...) 
or
2) Use a cyber parser tool: https://github.com/ApolloAuto/apollo/tree/master/modules/tools/record_parse_save  to convert this .record file data to .txt format. 
This idea seems easier to implement however I have also never done.
 
**I kindly ask if one of you have already done similar implementation? Or know some user from this community that have already done, if you remember about a similar question or this user, and could please assign to me this user or question. I will be very thankful.** 

Regarding my doubts about to use one of the 2 options described above. I will copy below from my original question on Apollo: https://github.com/ApolloAuto/apollo/issues/13178

DOUBT:

Hi @Yimags  Thanks very much for the complete answer, I got the idea, however if possible I would like to ask additional questions from my notes. Well let's go for the first option you told me:

**1 - Use a Listerner.py**: From this option I accessed this link https://github.com/ApolloAuto/apollo/blob/master/modules/tools/plot_control/plot_control.py

Looking at code I have noticed that it plots steering,brake and throttle data "online". I mean while I am running this node and the car is moving the plot is happening. I believe that this is the functionality of those graphics displayed online on Dreamview on PNC Monitor right? And **they are dynamic graphics**. Well I conclude if I wish other data (Odometry, GPS, Control, etc) I need to modify this code and implement my customized Listener.py correct? 

**2 - Cyber-record parser tool**

The second option seems a little bit more suitable for me, once I **do not wish "online dynamic graphics",** but to plot all Timestamp data from a window time (from beginning until the end of the simulation). So I would record all the channels from Apollo using the cyber record command. Afterwards as the tutorial you sent me says I can get the info of this recorded channels, and the output will be something similar to this:

[image]

This is good! Apollo records all channels, all of them contain messages! Awesome. **HOWEVER not all of them are PARSED by the parserrecord parser tool implemented in python**. As far as I have read the standard parsed data **are just related to the following sensors: radar, camera, lidar**. These are the standard sensors d**efined on parser_params.yaml** 

[image]

Therefore after executing the  $ record_parse_save command I will **get just .txt data from these sensors and not from the other channels, and this is bad!** I need the **other channels data converted to .txt** for further plotting this data for my master thesis discussion.

According to the tutorial I must change the code inside parser_params.yaml  if I wish the other data. Is this correct or is some other way to work around?  

If this be the only way than let me give an example to understand if this would work. Let me say I wish to parse data from pose and imu channels, instead those data from camera, radar and lidar . So the block of code instead radar, lidar and camera **would be (on parser_params.yaml) :**

```
pose:     # for position
  channel_name: /apollo/localization/pose
  out_folder_extn: ???
  timestamp_file_extn: pose_timestamp.txt

imu:     # for IMU
  channel_name: /apollo/sensor/gnss/imu
  out_folder_extn: ???
  timestamp_file_extn: imu_timestamp.txt
```

**Do you know to say what exactly The out_folder_extn means and how must I fullfill this area ?** 
Is Just the name of the folder where the data will be saved or extension of something that I don't know what? Extension of sensor data...Well I do not know how to rename this. Some Idea?

Sorry for the bunch of questions, but I really need to plot this data for my master thesis.

Let me know what would you do to get the other channels data parsed on .txt to further plot
and also about the 2 options (listener vs cyber record) would work better and how for all channels data?

Thanks in advance.

EricBoiseLGSVL, 2021-07-16:
I'll ping @hadiTab @cyberphysicalrobotics and @lemketron to see if they can help but to me it seems you should use option 2 and change code inside parser_params.yml  I am guessing the out_folder_exten is probably a folder path for the output folder so you can organize the data.  What does the other data types have?  An example of an existing one would help. Hopefully someone at apollo can let you know.

lemketron, 2021-07-16:
@hadiTab may have some suggestions but I would also suggest asking about this on the [Apollo issues](https://github.com/ApolloAuto/apollo/issues) page as you'll be able to talk with people more familiar with (and who work directly on) Apollo.

marcusvinicius178, 2021-07-16:
Yes I have already posted there this question, but no feedback was given. I will try again. In meanwhile I am going to wait the hadiTab hints. 
I tried the second option @EricBoiseLGSVL , however it seems the code is not properly working for the current path and folders the apollo 6.0 presents, considering I got the error below of file not found:

[image]

However the file is there as you can visualize below:

[image]

I do not know why this code didn't work...it should work and parse the data....

EricBoiseLGSVL, 2021-07-16:
do you need the full path?

marcusvinicius178, 2021-07-16:
I don't know what is going on actually on the code. The info from data record is always displayed correctly. But the code to parse this data to .txt format does not work, as you can check below:

[image]

From the error it is clear that some line of code to open the file is wrong! However it is quite hard for me understand this code and which line to change to point to the correct folder for the parser look into...some idea?

The record_parser_save.py module was not written by me. It was updated 6 months ago, maybe there is still some error...
https://github.com/ApolloAuto/apollo/blob/master/modules/tools/record_parse_save/record_parse_save.py

If you are a good programmer and could give me a hint on this regard I will be very thankful. I am still a beginner haha; Otherwise I will try to debug this code and see what happens.

Other question. Does the bridges working normally? I tried to edit today a simulation on the LG SVL last release, and I got some errors in the autopilots...

Thanks in advance

EricBoiseLGSVL, 2021-07-16:
Yes, the latest release has some issues with asset versions.  You may need to unlink cluster and relink. You also need to remove old vehicles and configs that point to old bridges.

As for code, I have a feeling you need to place the file in a certain folder that apollo expects.

marcusvinicius178, 2021-07-17:
Hi @EricBoiseLGSVL yes I needed to change the path to get the parsed data to the location my cloned repo was expecting. But this did not was the cause of the error. The code was updated and the tutorial which I followed on LG SVL website to run latest Apollo with LGSVL may contain old code....
However this was not enough to solve my issue. As I completely described and am answering here: 
https://github.com/ApolloAuto/apollo/issues/13948#issuecomment-881810003

The cyber recorder is good to record all the channels data on binary format. OK, Amazing! BUT the parser module provided, just parse data for the sensors: camera, lidar and radar as you can check:

[image]

Thus when I tried to parse the IMU recorded data, without having the imu_parser.py module I got obvisously an error:

[image]

Apollo did not create other modules to get parse the other data... and these modules are pretty big and hard to code....

Then I still ask myself and you guys other way to get these channels data and plot them? @hadiTab, @lemketron  have you already done some plot of some of the channels below?

[image]

Please let me know.

marcusvinicius178, 2021-08-02:
I have solved this issue and provided the tutorial here on my branch in apollo forked repo, if someone be interested:
https://github.com/marcusvinicius178/apollo/tree/extract_chan_data/cyber/python/cyber_py3/channels_data_extraction

I am going to create the code to convert csv to DataFrame and afterwards plot the graphs....then I will update this branche soon.

closed 2021-08-02 by marcusvinicius178