lgsvl/simulator #1477: Python-API/VSE based simulation on windows

tecena, 2021-06-07:
Hi,

I want to create own scenarios in the simulator and run. Either using VSE/ Python-API based with autoware AD-Stack . For this simulations to run do we need to setup the simulator environment on ubuntu? or can we run in windows also?

Currently I have setup simulator  in windows and autoware on ubuntu, I'm able to run the simulations with Random-Traffic template by setting the goal point in autoware RVIZ, But if I follow the same with VSE or Python-API based I'm getting error in simulations.

Please help
Thank you

EricBoiseLGSVL, 2021-06-10:
We are working on making VSE templates run on Windows but it is only Linux at the moment,

AKSHATA-MK, 2021-06-14:
> We are working on making VSE templates run on Windows but it is only Linux at the moment,

Thank you for the reply.
Can the Python API works in windows with autoware in ubuntu?

AKSHATA-MK, 2021-06-14:
> > We are working on making VSE templates run on Windows but it is only Linux at the moment,
> 
> Thank you for the reply.
> Can the Python API works in windows with autoware in ubuntu?

tecena, 2021-06-14:
> We are working on making VSE templates run on Windows but it is only Linux at the moment,

Thank you for the reply.
Can the Python API template work in windows with autoware AD stack in ubuntu?

lemketron, 2021-06-14:
> Can the Python API template work in windows with autoware AD stack in ubuntu?

Currently the VSE and Python API templates require the Simulator to run on Ubuntu. If you have Ubuntu for Autoware you might try running the simulator on Ubuntu as well to use these templates, or use the "API Only" mode on Windows and then you can run your python scripts from the shell on Windows.