lgsvl/simulator #1478: Problem about NPC vehicles

wobushigaoying, 2021-06-07:
Hello team,

After I run a simulation, if I did not move the ego vehicle, the NPC vehicles will not vanish and always travel following the annotated lanes. Once I moved the ego vehicle, the NPC vehicles will vanish If the NPC vehicle is out of sight ( or might exceed a preset distance away from the ego vehicle).

I went deep in the code, and I could not find what is going on.

Here are my issues:
 
I want to spawn 30 NPC vehicles in my scene, and they will not disappear wherever my vehicle is.

Another question: what do the following parameters mean? I am a little confused cus it is hard to notice the change between they are activated and not activated.

[image]

EricBoiseLGSVL, 2021-06-10:
If you set IgnoreNPCBounds to true, npcs will not check the NPCSpawnBounds distance.  Be aware this will effect performance.  The logic behind it is that when ego cannot "see" npc it should despawn.  If you want more control, I recommend using the python api and script the npc movement.

wobushigaoying, 2021-06-11:
Thanks for your tips. 

I am using LogitechG29 racing steering devices to drive the ego vehicle. I write a C# script and make it a prefab and then uploaded it to the cloud. Although Python API can do the same things, the simulation would be frustrating when I invoke simulation.run(time) even if I write some threads to collect G29 data and apply it to the ego vehicle. So, I prefer to adopt build-in achievement.

I followed your instructions to set ignoreNPCbounds to true, but the NPC vehicles will still disappear. I have uploaded my simulation screenshot. I modified MapOriginEdit.cs script to address more NPC vehicles in the scene. As the picture shows, take the left-half scene as an example, Lane1-Lane8 is a closed circle. I set DenySpawn to false on Lane2-Lane8 and true on Lane1. Thus the NPC vehicles will only spawn on Lane1. When the ego vehicle does not move, the NPC vehicles will spawn until they reach the maximum NPC count. The NPC vehicles far away from the ego vehicle will despawn when the ego vehicle starts to move. When the ego vehicle gets close to the first waypoint on Lane1, the NPC vehicles will spawn again.

I want to spawn 100 NPC vehicles in my scene, and they will not despawn wherever my vehicle is. I modified, adjusted, and tested some possibly related codes, but I still haven't solved it. Is there anything wrong with my Lanes or settings??  

[image]

EricBoiseLGSVL, 2021-06-11:
If you spawn 100 npcs you will crash the simulator.  We maxed out the npcs for performance at 30.  It is just too many pathing  calculations for more than that.

If you want you can edit source code in NPCManager to not despawn at all.