lgsvl/simulator #1479: Simulator crashed when launching it through remote server

ghostsun89, 2021-06-07:
My develop env is a Windows 10 desktop and a Linux 18.04 server with NV GPUs 1080Ti. I have launched LGSVL simulator successfully on the Linux. But the Linux server is not in my workplace, so I plan to remote the Linux using my desktop Windows10 through Xrdp. I have successfully done that, but when I launched the simulator remotely, the following problem occurred:
```
[Vulkan init] extensions: name=VK_NV_sample_mask_override_coverage, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_sm_builtins, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_subgroup_partitioned, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
[Vulkan init] extensions: name=VK_KHR_acceleration_structure, enabled=0
[Vulkan init] extensions: name=VK_KHR_ray_tracing_pipeline, enabled=0
Vulkan API version 1.1.0 (1.2.155 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 1080] id=[1b80]
Vulkan device type 2
Vulkan driverversion=[0x73124040] uint=[1930575936]
Vulkan PSO: cache data not found [path=/tmp/LGElectronics/SVLSimulator/vulkan_pso_cache.bin]
Begin MonoManager ReloadAssembly
- Completed reload, in  0.050 seconds
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 50 Hz
InitializeOrResetSwapChain 1024x768 hdr=0 samples=1
Caught fatal signal - signo:11 code:1 errno:0 addr:(nil)
Obtained 2 stack frames.
#0  0x007f62baabe980 in funlockfile
#1             (nil) in (Unknown)
</.config/unity3d/LGElectronics/SVLSimulator/Player.log" 185L, 12124C           185,1         Bot

```

My Linux Env:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.73.01    Driver Version: 460.73.01    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080    Off  | 00000000:02:00.0 Off |                  N/A |
| 28%   38C    P8     7W / 180W |    149MiB /  8118MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  GeForce GTX 1080    Off  | 00000000:03:00.0 Off |                  N/A |
| 27%   37C    P8     5W / 180W |      6MiB /  8119MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
```

The LGSvl Version I used is **svlsimulator-linux64-2021.1.1**

EricBoiseLGSVL, 2021-06-10:
Do you have mesa drivers installed?  Please remove if so.

Yuzhe-Tian, 2021-06-16:
Having the same issue here. With the mesa drivers and vulkan-dev all removed, the issue still exists.

tic-toc-developer, 2021-06-18:
I believe the problem lies within the XRDP service. [This page](https://askubuntu.com/questions/1231155/vulkan-driver-stops-working-when-running-in-google-remote-desktop) tells that Vulkan is current not compatible with XRDP, as XRDP renders another session of desktop.

EricBoiseLGSVL, 2021-06-21:
This would require changes on Unity's end.  We cannot make this change

closed 2021-07-16 by EricBoiseLGSVL