lgsvl/simulator #1483: Binary Options (not bundle)

NataliaSanchezSan, 2021-06-08:
I have a problem in the build simulator: "Binary Options (Not bundles)" and I can't add a New vehicle.

[image]

Yuchen413, 2021-06-09:
Same problem though have already added assets by git clone in External/environment/...

EricBoiseLGSVL, 2021-06-10:
You need to open the drop down menu.  See the arrow pointing right?  Click that.

EricBoiseLGSVL, 2021-06-10:
What do you mean by add a new vehicle? If you are building a binary, that is all you can do, build the binary.  If you are building a vehicle, environment, etc. you need to open the drop down and if you cloned the repo correctly, you will have a toggle to select one item to build a bundle. This bundle is built and placed in AssetBundles folder.  This must be uploaded to the web ui to be added to your simulation.  Please look at documentation [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).

Yuchen413, 2021-06-10:
> You need to open the drop down menu. See the arrow pointing right? Click that.

Thanks for your reply. I tried before but it didn’t work, there was no response when clicking the arrow.

NataliaSanchezSan, 2021-06-10:
If I select the build file:

[image]

But, I don´t know how to disable binary options.

NataliaSanchezSan, 2021-06-10:
Literally, it is not possible to select the vehicle, whether or not the build simulator is activated. 

[image]

EricBoiseLGSVL, 2021-06-10:
It says you are missing Golfinho.prefab.  Do you have the prefab in the root of the asset folder like the documentation says?

NataliaSanchezSan, 2021-06-10:
I created the prefab as indicated in the documentation in the assets / external / vehicles / Golfinho / Models / folder, I also tried to move Golfinho from the Models folder but it didn't work.

[image]

EricBoiseLGSVL, 2021-06-10:
Assets/External/Vehicles/Golfinho/Golfinho.prefab. Please also clone Jag repo to refer and see if you are missing class components on the vehicle prefab.

NataliaSanchezSan, 2021-06-20:
During this time I tried to add a new vehicle, with the advice you told me, but it was not possible.
So I decided to download the Jaguar repository, to check if any component is missing in the prefab, but it is not, I find everything correct, in addition the simulator still does not allow me to build the vehicle, in this case you cannot even select the Jaguar vehicle.

[image]

EricBoiseLGSVL, 2021-06-21:
Again, you have the prefab in the models folder.  Take the prefab and put in the root golfinho folder.

NataliaSanchezSan, 2021-06-22:
Thank you very much for your recommendations, the car is already integrated. I had to migrate to the Windows system because on Linux this integration was not possible.

closed 2021-06-22 by NataliaSanchezSan