lgsvl/simulator #1484: GPS location using python API

pushkar4501, 2021-06-09:
Hello, I am trying to get ego vehicle GPS location using Python API. I can already spawn ego and npc vehicles at my desired location. but how can i update gps value of the ego vehicle while running real time(using python api)
p.s. - i dont want to use rosbridge to collect the data for my project.

EricBoiseLGSVL, 2021-06-10:
We have created the bridge for data collection.  Python api is not performant for this task.  If you want to do this anyway, you can edit Api command scripts and the python api repo to collect the data.  We have examples already in the source code.  You will need to create a new custom binary.

pushkar4501, 2021-06-11:
Thank you for your reply. As i am a newbie here, could you provide me some references about creating a new custom binary for simulator.

lemketron, 2021-06-11:
Information about building the simulator from source can be found here:
https://www.svlsimulator.com/docs/installation-guide/build-instructions/

closed 2021-07-16 by EricBoiseLGSVL