lgsvl/simulator #1487: Shift to reverse gear does not work in SVL Simulator 2021.1.1

JuliaEmelianova, 2021-06-11:
I press 'page down' button in interactive mode in SVL Simulator 2021.1.1 for shifting to reverse gear, but it does not work. Forward drive gear remains be active

EricBoiseLGSVL, 2021-06-23:
Are you using the keyboard control sensor?

JuliaEmelianova, 2021-06-23:
@EricBoiseLGSVL yes, I use the keyboard control sensor

EricBoiseLGSVL, 2021-06-24:
Are you on Linux or windows OS?

JuliaEmelianova, 2021-06-25:
@EricBoiseLGSVL 
My system details are:
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo master (latest Apollo was cloned at 2021-06-05)
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

It seems that shifting to reverse gear works fine unless I run Apollo with SVL Simulator. As soon as a simulation is seen in the Apollo's Dreamview, shifting to reverse gear breaks, and EGO vehicle always drives forward. If I stop a Docker container, shifting to reverse gear works fine again.

lemketron, 2021-06-25:
> It seems that shifting to reverse gear works fine unless I run Apollo with SVL Simulator. As soon as a simulation is seen in the Apollo's Dreamview, shifting to reverse gear breaks, and EGO vehicle always drives forward. If I stop a Docker container, shifting to reverse gear works fine again.

Did you leave Apollo's "Control" module enabled?  If so then Apollo is controlling the gear and will always switch it back to neutral or drive. Try disabling the control module in Dreamview and see if you are able to switch to reverse manually.

JuliaEmelianova, 2021-06-25:
@lemketron Yes, Apollo's "Control" module was enabled. If I disable it, manual switching to reverse is possible. Thank you for the reply )

closed 2021-06-25 by EricBoiseLGSVL