lgsvl/simulator #1488: Problem how pedestrians move

krtt, 2021-06-11:
Pedestrians move too slowly, they sometimes cannot cross the street with green light, and they sometimes stop their walking animation (limbs movement) when moving around in the sim. Is there an option in SVL sim to speed up pedestrians? And ensure they cross the street when they are at a green light pedestrian crossing? And ensure their walking animation works correctly? Also, can their walking be made more directed, so that they are not wandering off during crossing the street?

blufissure, 2021-06-21:
hello,my pedestrian doen't walk,could you tell me how i can do to let the pedestrian walk? i  tried some behaviors,but that didn't work

EricBoiseLGSVL, 2021-06-23:
@blufissure please make a new issue, don't add to one created for a different topic.

@krtt yes you can change the animation controller to tweak when animations are transitioned.  We are still making changes to this and it is an issue when speed is low but not 0.  In pedestrian controller, you can edit the default speed or use api only mode to set the speed you want and waypoints.  During random simulation, there is no guarantee that a pedestrian will make the crossing or the path.  We use nav mesh agents for pathing.

krtt, 2021-06-27:
Where is the animation controller and how to tweak when the animations are transitioned? Same for the pedestrian controller.

EricBoiseLGSVL, 2021-06-28:
C:\Users\you\Documents\Repos\Simulator\Assets\Animations\PedestrianAC.controller

When you open and Animator Panel and select the animation controller, you get the animation state machine editor.  The arrows connecting the idle to the walk can be selected and in the inspector and the greater than less than values can be changed,  Instead of 0 use 0.01 etc.  See what values work best.  This is only transitions and animation speed.  You will need to edit pedestrian controller code to make the peds move faster.

closed 2021-07-16 by EricBoiseLGSVL