lgsvl/simulator #1492: Navigation breaks down and EGO does not drive in further tests when destination point under negative test is set

JuliaEmelianova, 2021-06-12:
**Description**
For negative test, destination point for EGO may be set "incorrectly" (e.g., drive into the opposite lane). In such a case, navigation breaks down and EGO does not drive in the further tests (see the attached video [TC2-12_2-15_2-16_BrokenNav.mp4](https://user-images.githubusercontent.com/80924057/121786009-584bdb80-cbce-11eb-93e1-9b5de8cad9be.mp4)) or its destination point is set improperly as the destination point from the last positive test. STOP and HOURGLASS signs occur near the EGO vehicle in Apollo's Dreamview window when navigation is broken.
It seems that there is something wrong with the Dreamview API method setup_apollo. If I set destination by using only setup_apollo method in my tests, navigation breaks down and EGO vehicle does not drive in further tests. If I set destination as setup_apollo with the initial position of EGO vehicle and then use Dreamview API method set_destination with the coordinates for negative test, then navigation does not break (however, sometimes it breaks also). In such a case, EGO stops and does not drive only in current negative test, further tests are not affected (see the attached video [TC2-12_2-15_2-16.mp4](https://user-images.githubusercontent.com/80924057/121785977-220e5c00-cbce-11eb-9bcc-4832c285d0fe.mp4)) .

**Steps to reproduce**
1. Run latest Apollo with SVL Simulator 2021.1.1
2. Select the "API Only" Simulation in SVL Simulator and click "Run Simulation"
3. Run the Python API script as unittest with three tests:
- _TC2-12_: EGO needs to turn left into the middle lane on a three-lane one-way road while being in the left lane with left and through mandatory movements on the two-lane one-way road (positive test)
- _TC2-15_: EGO needs to turn left into the middle lane with opposite motion on a three-lane one-way road while being in the left lane with left and through mandatory movements on the two-lane one-way road (negative test)
- _TC2-16_: EGO needs to turn right while being on the single-lane one-way road with the Stop line (positive test)

Code for setting the destination point for EGO is written as follows:
`

        dv = lgsvl.dreamview.Connection(sim, ego, BRIDGE_HOST)
        dv.set_hd_map('BorregasAve')
        dv.set_vehicle('Lincoln2017MKZ')
        dv.setup_apollo(egoX, egoZ, modules)

`
 where egoX, egoZ - destination coordinates under test.

**Expected results**
3 > - _TC2-12_: EGO turns left into the middle lane
      - _TC2-15_: EGO doesn't drive, navigation does not brake
      - _TC2-16_: EGO turns right

**Actual results**
3 > - _TC2-12_: EGO turns left into the middle lane
      - _TC2-15_: EGO doesn't drive, navigation is broken
      - _TC2-16_: EGO doesn't drive, navigation is broken

**Workaround (not always)**
Set destination point for EGO by using setup_apollo method with the initial position of the EGO vehicle. Then use set_destination method with the required destination coordinates under test.

Code for setting the destination point for EGO in such a case looks like:
`

        dv = lgsvl.dreamview.Connection(sim, ego, BRIDGE_HOST)
        dv.set_hd_map('BorregasAve')
        dv.set_vehicle('Lincoln2017MKZ')
        dv.setup_apollo(egoX_0, egoZ_0, modules)
        dv.set_destination(egoX, egoZ, egoY)

`
where egoX_0, egoZ_0 - coordinates of the initial position of the EGO vehicle,
egoX, egoY, egoZ - destination coordinates under test.

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo master (latest Apollo was cloned at 2021-06-05)
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1

ntutangyun, 2021-06-13:
when the destination fails, you need to check if the routing and planning module are still functioning. 

again maybe you should post this in Apollo repository

JuliaEmelianova, 2021-06-13:
Routing and planning modules are functioning during the whole test execution at step 3. It is seen in the attached [video](https://user-images.githubusercontent.com/80924057/121803284-cd142980-cc51-11eb-81c7-29a2cf784958.mp4).
Also I've opened issue in Apollo repository: https://github.com/ApolloAuto/apollo/issues/13850

JuliaEmelianova, 2021-06-15:
It seems that cause of the issue is in the routing and planning modules indeed. If I disable and enable these two modules at the end of each test case, issue is not observed. However, it is not clear why routing and planning modules were enabled all the time during my tests in Apollo's Dreamveiw window (also I checked their statuses via get_module_status method), but  destinaition failed

EricBoiseLGSVL, 2021-06-23:
We have noticed the need to check routing and planning.  Since you have an issue open in Apollo, can this issue be closed?

JuliaEmelianova, 2021-06-23:
@EricBoiseLGSVL yes, issue may be closed)

closed 2021-06-24 by EricBoiseLGSVL