lgsvl/simulator #1493: Navigation breaks down and EGO does not drive in further tests when the collision with sudden object occurs

JuliaEmelianova, 2021-06-12:
**Description**
When EGO collides with the sudden object (e.g., falling traffic cone or pedestrian), navigation breaks down and EGO does not drive in the further tests (see the attached video [TC2-1_2-4.mp4](
https://user-images.githubusercontent.com/80924057/121786818-68b28500-cbd3-11eb-948d-35293ce76fda.mp4
)) . STOP and HOURGLASS signs occur near the EGO vehicle in Apollo's Dreamview window.

**Steps to reproduce**
1. Run latest Apollo with SVL Simulator 2021.1.1
2. Select the "API Only" Simulation in SVL Simulator and click "Run Simulation"
3. Run the Python API script as unittest with two tests:
TC2-1: Some object falls on EGO at the beginning of motion. Different objects are tested: NPC vehicle, Pedestrian and Traffic Cone (negative test)
TC2-4: EGO needs to turn right while being in the right lane with right mandatory movements (positive test)

**Expected results**
3 > - TC2-1: EGO collides with the object and does not drive after collision
- TC2-4: EGO turns right

**Actual results**
3 > - TC2-1: EGO collides with the object and does not drive after collision, navigation is broken
- TC2-4: EGO doesn't drive, navigation is broken

**Comments**
The current issue may be related to the issue #1492

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo master (latest Apollo was cloned at 2021-06-05)
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1

ntutangyun, 2021-06-13:
you need to check if the planning module still functioning.. 

perhaps you should post this issue under Apollo repository.

JuliaEmelianova, 2021-06-13:
Routing and planning modules are functioning during the whole test execution at step 3. It is seen in the attached [video](https://user-images.githubusercontent.com/80924057/121804864-80cce780-cc59-11eb-9a7f-226118773d4f.mp4).
Also I've opened issue in Apollo repository: https://github.com/ApolloAuto/apollo/issues/13851
Thanks for your reply @ntutangyun

JuliaEmelianova, 2021-06-15:
It seems that cause of the issue is in the routing and planning modules indeed. If I disable and enable these two modules at the end of each test case, issue is not observed. However, it is not clear why routing and planning modules were enabled all the time during my tests in Apollo's Dreamveiw window (also I checked their statuses via get_module_status method), but  destinaition failed

closed 2021-06-23 by EricBoiseLGSVL