lgsvl/simulator #1494: Camera point of view in sim doesn't match sensor config randomly (second issue)

MaxandreOgeret, 2021-06-14:
Hello, 
I have made a new sensor configuration for the Lexus2016RXHybrid. And the camera look good in the web preview but the image I get from the simulator in ROS is clearly wrong. Here is my current config : 

[image]

<details>
  <summary><b>Click here to see YAML</b></summary>
      
```yaml
[
[
  {
    "params": {
      "Width": 1444,
      "Height": 1080,
      "Topic": "/camera_fl/image/compressed",
      "FieldOfView": 25.3
    },
    "transform": {
      "x": -0.17502,
      "y": -0.66163,
      "z": 1.11787,
      "pitch": -5.41366775,
      "roll": -0.3626822843,
      "yaw": 5.51836698
    },
    "name": "camera_fl",
    "parent": "Center Lidar",
    "pluginId": "3d4f1e08-4c62-4e9f-b859-b26d4910b85e",
    "plugin": {
      "assetGuid": "e8bb3908-0059-4829-8cc3-4ba07dd915bf",
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "id": "3d4f1e08-4c62-4e9f-b859-b26d4910b85e",
      "name": "Color Camera Sensor",
      "type": "ColorCameraSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This is the type of sensor that would be used for the Main Camera in Apollo.\nColor Camera also has multiple post processing sensor effects that can be added to the Postprocessing field in params. Effects can be combined with an array of Postprocessing fields but order is hard coded.\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#color-camera for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      }
    },
    "type": "ColorCameraSensor"
  },
  {
    "params": {
      "Topic": "/gps/imu",
      "Frame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
[... 404 lines cut ...]
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      }
    },
    "type": "GpsSensor"
  }
]
]
```
</details>

The problem seems to be random and can affect both cameras, sometimes separately, sometimes together, sometimes never.

Here's the view from the front left camera (bad) 
[image]

Here's the view from the front right camera (bad)
[image]

What could I do to solve this problem ? 
Thanks !

**P.S. Previous issue that has been closed [here](https://github.com/lgsvl/simulator/issues/1474).**

MaxandreOgeret, 2021-06-15:
It works with thisd yaml, even if it should change nothing. Seems that it is a bug.

```yaml
[
  {
    "params": {
      "Topic": "/gps/imu",
      "Frame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -0.4,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    },
    "name": "IMU",
    "parent": null,
    "pluginId": "9c34ec28-627d-4e73-9bb0-9c2aa7e978f5",
    "plugin": {
      "assetGuid": "d50ac8b1-0166-44a6-8499-5d8ecf7c8527",
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "id": "9c34ec28-627d-4e73-9bb0-9c2aa7e978f5",
      "name": "IMU Sensor",
      "type": "ImuSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This sensor output at a fixed rate of 100 Hz. IMU publishes data on topics where the 2nd topic has corrected IMU data.\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#imu for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      }
    },
    "type": "ImuSensor"
  },
  {
    "transform": {
      "x": 0,
      "y": 0,
      "z": -0.4,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    },
    "name": "Keyboard Car Control",
    "parent": null,
    "pluginId": "a2ff904a-ff06-4f06-9e45-cb58217a7142",
    "plugin": {
      "assetGuid": "ec968dc8-2e34-4307-8d3a-9bbffdf9e6cc",
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
[... 339 lines cut ...]
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "id": "3d4f1e08-4c62-4e9f-b859-b26d4910b85e",
      "name": "Color Camera Sensor",
      "type": "ColorCameraSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This is the type of sensor that would be used for the Main Camera in Apollo.\nColor Camera also has multiple post processing sensor effects that can be added to the Postprocessing field in params. Effects can be combined with an array of Postprocessing fields but order is hard coded.\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#color-camera for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      }
    },
    "type": "ColorCameraSensor"
  }
]
```

EricBoiseLGSVL, 2021-06-23:
Yes this is a bug, @thisisdavidfrank

EricBoiseLGSVL, 2021-06-23:
@MaxandreOgeret can you show your sensor config child parent relation in webUI please?  We might have fixed this already

MaxandreOgeret, 2021-06-24:
@EricBoiseLGSVL Here it is : 

[image]

EricBoiseLGSVL, 2021-06-24:
@MaxandreOgeret yes, I see now.  It is because of a bug with child objects.  Next release (testing at this moment :) ) This is fixed.  In the meantime you can move the sensor to the root base link.  Sorry about this.

closed 2021-07-16 by EricBoiseLGSVL