lgsvl/simulator #1496: Traffic signals fail if NPC vehicle is leading and follows waypoints

JuliaEmelianova, 2021-06-15:
When NPC vehicle is leading and follows waypoints (not lanes), traffic signals do not change on the crossroad.
This issue is observed for simulations created by means of both Visual Scenario Editor and PythonAPI Runtime Templates.
See the attached videos for issue observed in simulations created by means of Visual Scenario Editor Runtime Template:
[BrokenSignals_VisualEditor.mp4](https://user-images.githubusercontent.com/80924057/122120167-1ffd0500-ce3b-11eb-9380-46a16c45d1d7.mp4) - scenarios in Visual Editor are shown
[BrokenSignals_BorAve.mp4](https://user-images.githubusercontent.com/80924057/122120368-5cc8fc00-ce3b-11eb-9fa4-692483f26a90.mp4) - issue in simulation with map = BorregasAve
[BrokenSignals_SanFranc.mp4](https://user-images.githubusercontent.com/80924057/122120573-a580b500-ce3b-11eb-811d-b2c2de3fd6b7.mp4) - issue in simulation with map = SanFrancisco
There are two simulations with EGO and leading NPC when traffic signals fail and remain always red, and also two corresponding simulations without an NPC vehicle when traffic signals work fine and red color changes into green.

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo master (latest Apollo was cloned at 2021-06-05)
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

EricBoiseLGSVL, 2021-06-23:
@JuliaEmelianova waypoint mode does not work with traffic signals.  That is not supported, only random traffic

JuliaEmelianova, 2021-06-23:
@EricBoiseLGSVL Thanks for the reply. It works also when NPC vehicles follow closest lanes

closed 2021-07-06 by lemketron