lgsvl/simulator #1498: fisheye distortion

selfsimulator, 2021-06-17:
Hello, lgsvl team. 
I'm trying to extract the fisheye module of SVL simulator separately, so I deleted other codes that have little relevance. However, when the lensdistortion.cs and lensdistortion.compute files are identical, the image appears abnormal distortion. I also found other script data streams that are the same as those in SVL simulator. I will give abnormal pictures, hope to get your enthusiastic answer.
[image]
[image]

EricBoiseLGSVL, 2021-06-23:
Why are you extracting SVL distortion source code?

selfsimulator, 2021-06-24:
> Why are you extracting SVL distortion source code?

Because I just want to have an in-depth understanding of the fisheye realization process, But I didn't understand the distortion model and its principle of distortion calculation in "lensdistortion. Compute" and "lensdistortion. CS"！ So I would like to ask if there is any relevant article on the “calculation model of fish eye distortion” used by the simulator?
And thank you for your attentin on my problem!

EricBoiseLGSVL, 2021-06-24:
These issues will help #1047 #1009

selfsimulator, 2021-07-06:
Thank you again for your reply?
I have some questions for you.
The distortion parameter k1k2k3k4 in the document is reflected in the code as b1b2b3b4 calculated by a1a2a3a4. According to what calculation?
[image]
[image]
[image]
And I know that undistortedPos is the distortion polynomial in the document? The Parameter Xi is not specified in the document. What does this parameter mean? What is his geometric relationship with cubetexcoord`s XYZ?

selfsimulator, 2021-07-06:
> These issues will help #1047 #1009

Thank you again for your reply?
I have some questions for you.
The distortion parameter k1k2k3k4 in the document is reflected in the code as b1b2b3b4 calculated by a1a2a3a4. According to what calculation?
[image]
[image]
[image]
And I know that undistortedPos is the distortion polynomial in the document? The Parameter Xi is not specified in the document. What does this parameter mean? What is his geometric relationship with cubetexcoord`s XYZ?

lemketron, 2021-07-06:
@kamkolak-lge ?

selfsimulator, 2021-07-07:
> @kamkolak-lge ?

I mean, I want to know how to convert a1a2a3a4 to b1b2b3b4 in fisheye? And why do this? Where does its algorithm come from?

Then another question is what does the parameter "Xi" in distortion calculation mean? Where does the algorithm it participates in come from?

I didn't find any of the above explanations  in the document.

selfsimulator, 2021-07-14:
> @kamkolak-lge ?

This is my case code address“ https://github.com/selfsimulator/Fishey ”。 After it clicks the fish eye, the image quality will be particularly blurred, and then the cube box analysis is particularly large, and its splicing is wrong. But there is no change in the code algorithm of computeshader, so the cause of the error has not been found. What's more, I don't understand "undistortuvunified projection" in lensdistortion.compute. Where does the algorithm formula used come from?

closed 2021-07-14 by EricBoiseLGSVL