lgsvl/simulator #1499: Error Message Custom Sensor Plugin

digget, 2021-06-17:
Hey everybody, 

I tried to make a new Custom Sensor Plugin.
Build Progress was successfull. Then i uploaded it, attached it to the Vehicle Sensor Configuration and received following error message after i tried to start the simulation:

Error on latest run: Unloaded sensor type WheelSpeedSensor for GClass vehicle detected. Restart simulator application.

The i downloaded the ComfortSensor, build it unchanged but received the same error Message.
What am i doing wrong ?

Additonal information:
Build both with master branch in Unity Version 2020.3.3f1
I used it with Simulator Release 2021.1

EricBoiseLGSVL, 2021-06-23:
You have the sensor already loaded in editor.  Unity won't unload dlls at runtime so you need to restart editor and make sure sensor debug is disabled.

digget, 2021-06-23:
Hey Eric thank you for your answer.
I build my plugin in the unity editor, but used a already build version of the simulator which i downloaded from your homepage. I didnt use the Simulator inside the unity Editor. I haven't seen a option to disable sensor debug in your already build simulator.

The error Message didnt disappear, even after restarting

EricBoiseLGSVL, 2021-06-24:
Ah, I see.  What is the name of the sensor class? If you make a new sensor, you need to make a new class, you can't keep the same name.

digget, 2021-06-24:
The same error appears with a Plugin with another name.
I changed the Prefab name, the script in the prefab,
the name in the discription, filenames, class name. I changed the name everywhere, still the same issue.
What should i do ?

EricBoiseLGSVL, 2021-06-24:
- Ok, since you are using the binary (which version?) you do not have sensor debug mode, that is in Unity editor for using break points in sensors.
- Are you putting the sensor in the AssetBundles folder in the root of the binary? If so move it out.
- Open the binary and in the main menu top right, there is a drop down.  Click the Clear Cache button to open what is locally cached.  You probably have multiple sensors cached.  Delete all of them.
- Remove and delete the uploaded sensor to WebUI, remove from vehicle config and in your library.
- Re-upload the sensor and add to config again, run sim.

Hope this solves the issue.  Unity is not good with un-loading dlls.  It's the price we pay to forgo simulator dependencies ;)

digget, 2021-07-21:
Hey @EricBoiseLGSVL, 
i updated to 2021.2.

tried everything, this is the Error Message:
[image]

Tried to Debug, but i have no clue.

This is the Custom Message i want to publish, i want to publish a simple UInt32, i tried to name it UInt32 data.:
`
using UnityEngine;

namespace Simulator.Bridge.Data
{
    public class TickData
    {
        public uint data;
    }
}
`

My first approach was to add a Struct to Ros.cs to Publish a std_msg/UInt32.

EricBoiseLGSVL, 2021-07-22:
Did you build the bridge with your new data type?  You may need to edit simulator source as well in RosConversions.cs @hadiTab ?

digget, 2021-07-22:
There is no RosConversation in my Source code. Is there any Tutorial how to do it ? Should i Download the ros github repository ? I added the TickData to the Simulator.Data Namespace and the Folder. What should i do and how ?

EricBoiseLGSVL, 2021-07-22:
RosConversion.cs and yes it is in source code.  Yes, you will also need to download the ROS github repo since you want to add this data type.  I recommend waiting until we release 2021.2.1 because we have an issue with custom bridges.  In the meantime follow the current RosConversions.cs code to see how a data type is added.  You need the bridge to be able to send the serialized data and parse it.

digget, 2021-07-22:
Okay, for may basic understanding:
I download the ROS repository to Externals -> Bridges.
Then i add my stuff to the downloaded repository.
And then ?
How do i use my custom bridge ?

EricBoiseLGSVL, 2021-07-22:
You need to build the bridge, upload to webui and assign your vehicle to use this new bridge in the simulation.

digget, 2021-07-22:
First of all thank you for your time.
Second, i build the bridge plugin, but how can i select it ?

[image]

there is only the option for "other ROS Autopilot"

and another question, hats the difference between the Ros source code which comes with the source code of lgsvl and the extra ROS repository ?

EricBoiseLGSVL, 2021-07-23:
What type of simulation are you trying to run?

digget, 2021-07-23:
A simulation of an autonomous rc car. Therefore i use our custom autosoftware stack, which runs with ROS. An im trying to build a wheel speed sonsor plugin, to simulate a wheel encoder.

EricBoiseLGSVL, 2021-07-23:
Sorry I mean what runtime template?  Did you upload the bridge to your plugins in the webui?

lemketron, 2021-07-23:
> A simulation of an autonomous rc car. Therefore i use our custom autosoftware stack, which runs with ROS. An im trying to build a wheel speed sensor plugin, to simulate a wheel encoder.

I hope you'll consider sharing the sensor publicly (built and/or source code) if it is a common sensor that might be useful to others... thanks!

digget, 2021-07-23:
Finally, i found it

[image]

I searched for weeks, how to build a custom bridge and how to select a custom bridge.
I havent tested it yet, im sooo glad that i found the button in the sensor configuration. 
Pleace, by the name of God change that menu.

EricBoiseLGSVL, 2021-07-23:
@digget lol I'll pass that feedback to the web team @thisisdavidfrank .  Glad you got it working.  This is the first release of custom bridges and we still have some issues to work out.  Thanks for posting and please state your solution if you can for others, thanks.

digget, 2021-07-23:
To clarify, the Simulator is very nice and the web ui also. But the custom bridges option, catastrophic.

EricBoiseLGSVL, 2021-07-23:
Yeah, we are still figuring out the best UX for these new features.  Do you have any suggestion on how you would like the UX? Any feedback is helpful.

digget, 2021-07-29:
It would be nice, if its less hidden, little bit more intuitive
Something like that:
[image]

EricBoiseLGSVL, 2021-07-29:
Thanks for the feedback and using the simulator.  @thisisdavidfrank

closed 2021-07-29 by EricBoiseLGSVL