lgsvl/simulator #1500: Ego vehicle goes and stops alternately when control vechicle by  autoware topic /vehicle_cmd

blacksheep0303, 2021-06-17:
Hi, thanks in advance for any reply.
If this issue is not suitable for this repository, please inform and close it, thanks!

I set up a simulation environment which connect Lgsvl and Autoware.AI1.14 .Then I send topic /vehicle_cmd to control ego vehicle.
There are two mode to control ego vehicle: twist_cmd or ctrl_cmd .
When I send /vehicle_cmd { ... gear_cmd: {gear: 64} ... ctrl_cmd: {linear_velocity: 1.0, linear_acceleration: 1.0, steering_angle: 0.0} } , the ego vehicle goes well.

However,when I send /vehicle_cmd { ... gear_cmd: {gear: 0} ... twist:{linear: {x: 1.0, y: 0.0, z: 0.0}  angular: {x: 0.0, y: 0.0, z: 0.0}} , the AD Accel Input is sometimes more than 0 and sometimes less than 0.It makes ego vehicle go and stop alternately instead of moveing foward at  given speed (which is x:1.0).
I can't understand why the AD Accel Input is less than 0 when the speed is less than 1 .This makes ego vehicle's  speed vary between 0 and 1 , which means that the vehicle is running very unsmoothly.
Here are some pictures of AD Accel Input.
[image]
[image]
[image]
[image]

Can anyone help me with this question? Why ego vehicle always brake when I use twist_cmd to control the movement of the vehicle.Thanks a lot!

EricBoiseLGSVL, 2021-06-23:
Negative means that the brakes are being applied.  Not sure why twist would differ.  @hadiTab ?

rythm07, 2021-08-11:
I'm having the same problem. Anyone can help? @hadiTab @EricBoiseLGSVL