lgsvl/simulator #1501: Why the width of lane cannot be changed?

Addi2020, 2021-06-17:
I changed the Display lane width in inspector tab while doing map annotation but when i export base_map.bin file and upload it to dream view, It always display the same width (by default = 3.5). 

Is it wrong way to change the width of lane?

EricBoiseLGSVL, 2021-06-23:
The exporter defaults to 3.5. See line 547 in ApolloMapTool.cs.  You could extend the code to use this value in apollo export.  We will support sometime soon.

Display Lane Width is just a visualizer in map.  Lanelet2 is the only exporter that uses this value.

Addi2020, 2021-06-24:
where do I find ApolloMapTool.cs.? in apollo code?

ntutangyun, 2021-06-24:
> where do I find ApolloMapTool.cs.? in apollo code?

Assets/Scripts/Editor/ApolloMapTool.cs

Addi2020, 2021-06-25:
@EricBoiseLGSVL Thanks. Its working great to change the width of lane. 

But I have another issue that it changes the width of all the Maplanes together. Is there any way to change width of only one Maplane separately?

EricBoiseLGSVL, 2021-06-26:
I recommend a new property in the class and you write code to use this new property.  Without seeing what you changed, it is hard to tell.

closed 2021-07-29 by Addi2020