lgsvl/simulator #1502: Radar with ROS1 bridge : NullReferenceException:

MaxandreOgeret, 2021-06-18:
Hello, I try to use the radar with the ROS1 bridge and I get this error in the player.log : 

```
NullReferenceException: Object reference not set to an instance of an object
  at Simulator.Bridge.Ros.Conversions.RosConvertFrom (Simulator.Bridge.Data.DetectedRadarObjectData data) [0x000f7] in <1d0e33dd01be444d90c7c889462e0a7f>:0 
  at Simulator.Bridge.Ros.RosBridgeFactoryBase+<>c__DisplayClass2_1`2[DataType,BridgeType].<RegPublisher>b__1 (DataType data, System.Action completed) [0x00000] in <1d0e33dd01be444d90c7c889462e0a7f>:0 
  at Simulator.Bridge.BridgeInstance+<>c__DisplayClass22_0`1[DataType].<AddPublisher>b__0 (DataType data, System.Action completed) [0x0004f] in <0dbf30fcd0354901964b79d265b73453>:0 
  at Simulator.Sensors.RadarSensor.Update () [0x000b4] in <2ba58a4d9ece4139aa2348d54ee0e4e9>:0 
 
(Filename: <1d0e33dd01be444d90c7c889462e0a7f> Line: 0)
```

I believe the sensor is supported with the ROS1 bridge because the function `RosConvertFrom(DetectedRadarObjectData data)` is defined here : https://github.com/lgsvl/simulator/blob/fdea5ad112a02e24c06dac8c934328ec139d0763/Assets/Scripts/Bridge/Ros/RosConversions.cs#L414

Here's the definition of the radar sensor : 

```yaml
{
    "params": {
      "Topic": "/radar_fc",
      "Frame": "radar_fc"
    },
    "transform": {
      "x": 0,
      "y": 0.07,
      "z": 3.6,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    },
    "name": "Radar Sensor",
    "parent": null,
    "pluginId": "215b7d1e-7ad3-4fe6-b3d9-db0e229eb0a4",
    "plugin": {
      "assetGuid": "68ddcc86-1b72-447e-8b48-736917e07281",
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "id": "215b7d1e-7ad3-4fe6-b3d9-db0e229eb0a4",
      "name": "Radar Sensor",
      "type": "RadarSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This sensor outputs the objects detected by the radar. Detected objects are visualized with a box colored by their type:\n\t[\n\t\t\"Car\": \"Green\"\n\t\t\"Bicycle\": \"Cyan\"\n\t\t\"Agent\": \"Magenta\"\n\t]\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#radar for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      }
    },
    "type": "RadarSensor"
  },
```

How could I solve this problem ? 
Thanks for your help.

MaxandreOgeret, 2021-06-18:
solution : https://github.com/lgsvl/simulator/issues/1085#issuecomment-805546952

closed 2021-06-18 by MaxandreOgeret

bruce-almon, 2021-07-08:
> solution : [#1085 (comment)](https://github.com/lgsvl/simulator/issues/1085#issuecomment-805546952)

That is the solution for ros2
My problem is that when I do rostopic echo /radar_fc, nothing is printed.
Did you have the same problem?

MaxandreOgeret, 2021-07-09:
Worked for ROS 1.