lgsvl/simulator #1505: Ubuntu 20.04 - No Simulator menu on top toolbar

kk2491, 2021-06-18:
Hi Guys,

Good Day.

Ubuntu version : 20.04   
LGSvl Simulator Git branch used : Master
Unity Hub version : 2019.4.18f1 

I am trying to build SVLSimulator from source on Ubuntu 20.04. I am following the documentation given [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/#installing-unity-editor).  

I am not getting the **Simulator** option in the Unity editor. 

1. Installed Linux support and Windows support in Unity hub. 
2. As per the documentation I made sure Git LFS is initialized before cloning the repository.
3. Added the project by selecting the cloned repository directory.   
[image]
4. This is the Unity editor screen I get once I open the project.   [image] 
5. There is no **Simulator** option in the Unity editor.
6. Console.  
[image]
7. Project.  
[image]

I have tried the suggestions given in [Issue361](https://github.com/lgsvl/simulator/issues/361). However, I am still not able to fix the issue.

It would be great if you could guide me in the right direction. 

Thank you,  
KK

EricBoiseLGSVL, 2021-06-23:
please check that you have installed git lfs before you cloned.  There are many issues here if you have questions. To check read the documentation [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/). verify that you have image files and not text files.

kk2491, 2021-06-24:
Hi @EricBoiseLGSVL  Thanks for your reply. 
I followed the documentation step by step. And also made sure I have installed ```git lfs``` before cloning the LGSvl repository. 
Not sure what is going wrong here. 

Thank you

EricBoiseLGSVL, 2021-06-24:
I see you are on master.  This is a preview release branch for users that are familiar with simulator already and want to preview new features not released yet.  The version for master unity is 2020.3.3f1, you should see this in Unity Hub when you open the project. Never change the version of Unity for the branch, this will break... everything :). You are using the wrong unity version.  Please only use official release branches.  release-2021.1.  That will work with 2019.4.18f1 and with documentation.  I would re-clone because Unity fails when you downgrade a project version.  Sorry about the confusion on master, we will improve our messaging on this.

kk2491, 2021-06-24:
Thank you so much for the detailed description. Let me give a quick try and come back. 
Please keep the issue open until then.

kk2491, 2021-06-25:
@EricBoiseLGSVL Thanks, your suggestions helped me to get the Simulator up and running. 
As you suggested I used 2020.3.3f1 and found the ```Simulator``` option in the menu. 

Thanks again for your help.

EricBoiseLGSVL, 2021-06-26:
Great, my post states to use official release tags but glad it worked.  FYI master is not stable so we don't recommend development on this branch.  It will make it easy to use our new release out soon though ;)

closed 2021-06-26 by EricBoiseLGSVL