lgsvl/simulator #1506: Support for LGSVL Simulator 2020.06 Version

QZJGeorge, 2021-06-21:
Hi,

I have been working on LGSVL Simulator 2020.06 version for a few months. However, recently I am unable to connect to the old LGSVL server. I was not able to sign up or sign in and could not load the localhost: 8080 web page. Is the server shut down due to the new SVL Simulator release? I have tried the new release but found the 2020.06 version more open and more stable. Is there anything I can do to make the 2020.06 version work? Thanks!

[image]

iKoltcov, 2021-06-21:
Have the same issue, I tried to change `cloud_url` in the config.yml to `https://wise.svlsimulator.com/`, and looks like not route was changed only, but and the contract also.

EricBoiseLGSVL, 2021-06-24:
Sorry, we had an issue with the server and space but should be up and running now.  Let us know if you are still having issues.  @iKoltcov wise.svlsimulator.com is not for 06 and will not work, sorry.  Cloud is for all later releases.