lgsvl/simulator #1507: Why Error on latest run: BundleFormat version mismatch

DZiYou, 2021-06-21:
When I modified the appearance color of jaguar2015xe, I recompiled it through unity and added it to SVL simulator. But there was an error starting the simulation
Error on latest run: BundleFormat version mismatch
I want to know why?
 thank you

EricBoiseLGSVL, 2021-06-23:
What version of simulator are you using? What commit on of Jaguar repo are you using?

DZiYou, 2021-06-24:
Thanks for your reply。
The simulator I use is svlsimulator-linux64-2021.1.1. The car model has been released to the store, and its name is jaguar2015xe_ Blue, ZY Dou

DZiYou, 2021-06-24:
[image]

NataliaSanchezSan, 2021-06-24:
I also have this problem, I have already managed to place the vehicle, but it won't let me run the simulation.

EricBoiseLGSVL, 2021-06-24:
Did you build the vehicle in Simulator repo master and try to run on 2021.1.1 release?  If so, this is why you have a bundle mismatch.

DZiYou, 2021-06-25:
What does Simulator repo master mean? I build it by the following step:
[image]

DZiYou, 2021-06-25:
Is it because the jaguar2015xe model I downloaded is 2019 and the simulator is 2021.1.1? How to upgrade?

lemketron, 2021-06-25:
> Is it because the jaguar2015xe model I downloaded is 2019 and the simulator is 2021.1.1? How to upgrade?

Quite possibly...

If you are looking for the source to build that vehicle for SVL Simulator 2021.1.1 then you should check out the `2021.1` tag of the `Jaguar2015XE` repository here: https://github.com/lgsvl/Jaguar2015XE/tree/2021.1

And of course, make sure you are using the `2021.1.1` tag of the SVL Simulator project as well and NOT `master` (which is newer, and not compatible) to build the vehicle asset bundle: https://github.com/lgsvl/simulator/tree/2021.1.1

EricBoiseLGSVL, 2021-06-26:
@DZiYou yes, you are on 2020.3.3 and that is preview/master only.  You cannot use older versions of assets on this branch.  Please use a release tag of source for both simulator and asset repos to be sure you have the correct version of assets. Master is for preview of up and coming features and is not stable.

closed 2021-06-26 by EricBoiseLGSVL