lgsvl/simulator #1509: Cannot properly run Python Test Case Simulation

krtt, 2021-06-21:
When I follow the Creating and running a Python Test Case Simulation tutorial [here](https://www.svlsimulator.com/docs/running-simulations/running-simulator/#pythontestcasesimulation), two problems occur when I am at step 9:

1. error "AD stack is not ready after 30 seconds"
[image]

2. in Apollo Dreamview the map is not shown, even if it is downloaded and made available using this tutorial [here](https://www.svlsimulator.com/docs/system-under-test/apollo5-0-instructions/#adding-an-hd-map), and the map file is in the correct location in apollo root (/modules/map/data)
[image]

OS: Kubuntu 18.04

krtt, 2021-06-21:
When I try another test-case, with a map that is already available by default in Apollo, then the result is in error "AD stack is not ready":
[image]

Where it is possible to see more details what is the root cause for this?

EricBoiseLGSVL, 2021-06-23:
Please post your player.log from simulator to see what errors are occurring.

krtt, 2021-06-28:
As the old player.log is not useful any more after many different runs, then I tried to reproduce the problem from scratch. But as Apollo perception module has stopped working suddenly, I tried to run a Python Test Case Simulation with the modular testing option (which I have not tried before). The error is different now, and it still does not run.

Error:
`Error on latest run: Test case exit code: 1 error data: 2021-06-28 09:15:42,253 [INFO][__main__] Run python script scenario.py Traceback (most recent call last): File "scenario.py", line 74, in <module> dv = lgsvl.dreamview.Connection(sim, ego, BRIDGE_HOST) File "/usr/local/lib/python3.6/site-packages/lgsvl-0.0.0-py3.6.egg/lgsvl/dreamview/dreamview.py", line 36, in __init__ File "/usr/local/lib/python3.6/site-packages/websocket_client-0.57.0-py3.6.egg/websocket/_core.py", line 515, in create_connection websock.connect(url, **options) File "/usr/local/lib/python3.6/site-packages/websocket_client-0.57.0-py3.6.egg/websocket/_core.py", line 223, in connect options.pop('socket', None)) File "/usr/local/lib/python3.6/site-packages/websocket_client-0.57.0-py3.6.egg/websocket/_http.py", line 121, in connect sock = _open_socket(addrinfo_list, options.sockopt, options.timeout) File "/usr/local/lib/python3.6/site-packages/websocket_client-0.57.0-py3.6.egg/websocket/_http.py", line 201, in _open_socket raise err File` ...

Player.log attached.
[Player.log](https://github.com/lgsvl/simulator/files/6727024/Player.log)

If you could also guide how to read out useful information from that log in this case, would be great. Thanks!

EricBoiseLGSVL, 2021-06-28:
Looks like you don't have the SF map loaded in dream view or the name is not right.
```
[PROC][4819][ERR] HD Map SanFrancisco was not set. Verify the files exist in /apollo/modules/map/data/SanFrancisco and restart Dreamview -- Aborting...
6/28/2021 12:10:00 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2:0|WebSocketSharp.WebSocketException: The header part of a frame could not be read.
                              at WebSocketSharp.WebSocketFrame.processHeader (System.Byte[] header) [0x0001f] in <84d0020558774b6c88b3203b756a09ec>:0 
                              at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <84d0020558774b6c88b3203b756a09ec>:0 
                              at WebSocketSharp.Ext+<>c__DisplayClass59_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x0008e] in <84d0020558774b6c88b3203b756a09ec>:0  
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
```
It's tough to read the player.log and you get better as you read thousands of them :)  I always scan for any null errors or [ERR].  After you find one the rest of the log is not worth reading.  Once unity encounters a null or the [err] all other errors are false positive.

Check that the SF hd map is loaded in dreamview and or the file name is wrong.

closed 2021-09-28 by krtt