lgsvl/simulator #1510: 504 Gateway Time-out while visit localhost:8080

chenxinxu333, 2021-06-22:
While I start lgsvl (2020.06) in ade environment.  I met "504 Gateway Time-out" in my Chrome browser. I also cannot visit the url “
account.lgsvlsimulator.com”.
[image]

The same situation like https://github.com/lgsvl/simulator/issues/1217 . Does it mean the upstream server went down?

michael2012z, 2021-06-22:
Same to me. The web site has been inaccessible since yesterday.

chenxinxu333, 2021-06-22:
Now, it works again.

iKoltcov, 2021-06-22:
> Now, it works again. No timeout.

2 hours ago it was working but now there is an error again

chenxinxu333, 2021-06-22:
> > Now, it works again. No timeout.
> 
> 2 hours ago it was working but now there is an error

chenxinxu333, 2021-06-22:
Now, it still works. @iKoltcov. I found that once you visit the localhost:8080 successfully, there are some cache in your browser. And if you enter ade environment in the same adehome direcotory, you can visit localhost:8080 again and start lgsvl successfully even if the upstream server went down.

EricBoiseLGSVL, 2021-06-23:
Sorry, we had an issue but it should be resolved.

closed 2021-06-23 by EricBoiseLGSVL