lgsvl/simulator #1512: intersection laneline problem

honey5967, 2021-06-23:
[image]
There is a problem that the lane mesh inside the intersection is not created in the simulator. Also, only the line width and spacing defined in "Hd Map Mesh Builder" are applied, not the lane width or type defined in the OpenDRIVE file. On this matter I don't know...
I want to create additional Objects models, can I add a .fbx file with the same name as the OpenDRIVE objects name to the simulator resource?
Thank you..!

honey5967, 2021-06-23:
A lane at an intersection is not a lane belonging to OpenDRIVE lanes, but a vehicle guideline represented by objects.

honey5967, 2021-06-23:
It wasn't about the intersection. I'm sorry

EricBoiseLGSVL, 2021-06-23:
@honey5967 no worries, I think I get what you mean.  HD Map Mesh builder is just to help users that don't have access to 3d modeling tools.  We don't support OpenDrive lane width at this time, sorry.  You can create/add any objects you want to the scene, just be sure to add all assets in the Assets/External/Environments/YourMap folder so they are bundled correctly and save the scene.  Look at CubeTown repo to see how the scene is organized in the folder and in the unity scene.  Have fun creating and let us know if you need help.

honey5967, 2021-06-24:
Thank you..! curiosity has been resolved

closed 2021-06-24 by EricBoiseLGSVL