lgsvl/simulator #1513: How to Enable Lane-line Detector

joeking11829, 2021-06-24:
Hi LGSVL Teams,

According to this document : [Lane-line Detector](https://www.svlsimulator.com/docs/simulation-content/lane-line-detector/)

It mean that we can add lane-line on segmentation camera. 
[image]

When I followed the document's steps.
I can create the "LineData_generated".

After compiled the simulator, these is no lane-line on segmentation camera.
What did I miss up ?

Thanks.

EricBoiseLGSVL, 2021-06-24:
Did you create the map bundle and then add the lane line sensor to the vehicle?

joeking11829, 2021-06-25:
> Did you create the map bundle and then add the lane line sensor to the vehicle?

Thank you for your reply.

1. Sorry, I'm not familiar with Unity. 
    "create the map bundle" -> Do you mean that I should rebuild the map which I used in LGSVL ?
2. I didn't add lane-line sensor, so if i add the lane-line sensor then I should get the lane-line information from the segmentation sensor ?

EricBoiseLGSVL, 2021-06-26:
You need to create a bundle with the new data for lane lines, see docs [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).  Yes, you need the sensor to display the lane lines in the visualizer.

joeking11829, 2021-06-29:
> You need to create a bundle with the new data for lane lines, see docs [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/). Yes, you need the sensor to display the lane lines in the visualizer.

Thanks for your tips.
I want to add the lane lines to the "San Francisco" map, What should I do ?

EricBoiseLGSVL, 2021-06-29:
- Run the create lane lines from editor for SF
- Save scene
- Create a new bundle
- Upload to webUI.
- Rename to SFwithLL or something
- Add LaneLineSensor to the vehicle sensor configuration
- Create simulation with the new map and vehicle config
- Run simulation and open lane line visualizer in the simulator UI to see what is shown in documentation

Let us know if you have any issues

joeking11829, 2021-06-30:
Thanks for explain the detail steps.

I can reproduces the lane-line for CubeTown successfully.
How can I find the "San Francisco" repo ?

thanks again for your help

EricBoiseLGSVL, 2021-06-30:
Sorry looks like SF repo got accidentally set as private.  We will fix soon.  Thanks for letting us know.

joeking11829, 2021-07-01:
Okay, I got it.
Thank you.

closed 2021-07-01 by joeking11829