lgsvl/simulator #1515: Semantic Cameras not rotated

ToniHille, 2021-06-25:
Hello, 
I am using the newest Release and I have a car with 6 semantic cameras:
[image]

The camera in front is not rotated, the two cameras left have a yaw of -90 degree and the two to the right 90 degree. The camera looking backwards has a yaw of 180 degree.

I use the Python API save() method for all cameras. 
My expectation would be, that I have a  360 degree FOV when combining the pictures (Each camera looks in the direction in which it was rotated). But it seems all images are just looking straight into z direction without any rotation. The translation is applied.

Examples:
* Front Center: 
[image]
* Rear Center: 
[image]
* front right: 
[image]

I will attach my config json file as txt
[six_semantic_cameras.txt](https://github.com/lgsvl/simulator/files/6716320/six_semantic_cameras.txt)

Could you please help me resolve this? What am I doing wrong?

EricBoiseLGSVL, 2021-06-26:
@zelenkovsky is z forward live in wise preview?

ToniHille, 2021-07-01:
Hello, are there some updates?
I did some more investigate and when doing the same with a normal color camera sensor, everything seems fine. The backwards camera image is looking backwards as expected.

Maybe a other hint, that something is wrong with the semantic camera plugin -> The sensor name is not updated.
In the GUI i named all semantic cameras different, like "Rear Left", "Rear Right" and so on. But when I call 

`for sensor in ego_vehicle.get_sensors():
      print(sensor.name)`
I get the following output:
```
Front Color Camera
Segmentation Camera
Segmentation Camera
Segmentation Camera
Segmentation Camera
Segmentation Camera
Segmentation Camera
Segmentation Camera
Segmentation Camera
```
As you can see, for the color camera it is working ("Color Camera Sensor" would be the default) but for the semantic all have their default name value.

Any ideas or hints?

EricBoiseLGSVL, 2021-07-02:
Sorry we have not seen this issue yet.  @revati-naik can you make a bug report to look into this please?  We have many fixes in webui coming in next release and this may be already resolved.

ToniHille, 2021-07-05:
Thank you, please let me know if you need any more information to reproduce the bug.

kamkolak-lge, 2021-07-06:
@EricBoiseLGSVL This has been fixed on internal repo, in `03af43863c28926562bbeb12db4f5c052f3f280b`

@ToniHille The issue is related to cubemap rendering on sensors with custom pass. Until the fix goes public in next release, I suggest to disable fisheye effect on segmentation sensor (it's the effect requiring cubemap). You should still be able to use standard distortion.

ToniHille, 2021-07-08:
Hello, thank you for the clarification and fast answer.

closed 2021-07-08 by EricBoiseLGSVL