lgsvl/simulator #1516: ROS2 bridge disconnects on 2021.1

danielm1405, 2021-06-25:
I am trying to run SVL 2021.1 with Autoware.Auto on Ubuntu 20.04 with the following assets:

[image]

The simulator launches but the bridge does not work properly. "Bridge status" appears as "Connected" for a fraction of a second, some messages counters go up, but then it disconnects for a few seconds and the behavior repeats.

https://user-images.githubusercontent.com/39591259/123428779-7817d200-d5c6-11eb-9fd9-d3c50b15c872.mp4

I tried to run SVL 2021.1 both from prebuilt binary you provide and built from sources by myself. The behavior remains the same.

Am I doing something wrong? Any help would be appreciated.

EricBoiseLGSVL, 2021-06-26:
Please post the player.log to see what the error is, thanks

danielm1405, 2021-06-28:
Hi, I discovered this docs https://www.svlsimulator.com/docs/system-under-test/autoware-auto-instructions/ and were able to run it properly. 

The issue was that I didn't have a bridge running at all! I assumed that it was launched automatically as the "Bridge status" was changing between "Connected" and "Disconnected". The statuses are very confusing to me. I don't have a clue why msgs counter were going up with bridge missing.

I attach my Player.log to enable further investigations: [Player.log](https://github.com/lgsvl/simulator/files/6726696/Player.log)

EricBoiseLGSVL, 2021-06-28:
Glad you found the issue.  Yes, you need to run the bridge.  The msg will still try to send.

You also may have an outdated/duplicate sensor you need to remove from Clear Cache in main menu.  It also says that is has been already loaded.  Unity can't unload dlls at runtime.  I recommend deleting all sensors in clear cache esp IMU.
```
failed to load asset from 88c08315-d1e5-4dea-9970-2f477ac63553/manifest.json: Object reference not set to an instance of an object STACK:   at Simulator.Utilities.SensorTypes.GetConfig (System.Object sb) [0x00006] in <15ed6a56843d489d84c2a18a0305758a>:0 
  at Simulator.Web.Config.LoadSensorPlugin (Simulator.Manifest manifest, VirtualFileSystem.VfsEntry dir) [0x001d1] in <15ed6a56843d489d84c2a18a0305758a>:0 
  at Simulator.Web.Config.CheckDir (VirtualFileSystem.VfsEntry dir, Simulator.Web.Config+AssetLoadFunc loadFunc) [0x0006a] in <15ed6a56843d489d84c2a18a0305758a>:0  
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
```

I recommend clearing cache often because of the limitations of Unity to unload while you develop.  We are still working on ways to make messaging better.

danielm1405, 2021-06-29:
Are you referring to GI Cache (_Edit -> Preferences -> GI Cache -> Clean Cache_)?

EricBoiseLGSVL, 2021-06-29:
No, in simulator main menu when you start the application, top right drop down.

closed 2023-11-24 by danielm1405