lgsvl/simulator #1517: Use the lidar sensor script in a project

manone888, 2021-06-25:
Hi, I am creating a simple project in Unity that uses a lidar sensor to detect objects and publishes the point cloud in ROS message.

I would like to know if there is a way to use your lidar sensor script to create a lidar sensor in my project and visualize the point cloud?

Could you please advise me on this? It would be appreciated if you could point out any related documents. 

Thanks a lot!

EricBoiseLGSVL, 2021-06-26:
Depends, @lemketron what is the stipulations of the license?

lemketron, 2021-06-27:
> I would like to know if there is a way to use your lidar sensor script to create a lidar sensor in my project and visualize the point cloud?

Technically it's certainly possible, but the LiDAR sensor and ROS bridge code likely rely on may other parts of the simulator core project; I'm not sure how separable they are, and of course the other issue is the license. While we're happy to openly share the source code for our simulator, our intention was for users to build on top of what we've done and contribute to our growing community by publishing new (sensor, bridge, and other) plug-ins as well as sharing bug fixes and other enhancements to our simulator.

Refer to section 1.4 of the [LICENSE](https://raw.githubusercontent.com/lgsvl/simulator/master/LICENSE), which indicates that your use of derivative works require a separate license agreement:

> 1.4	Derivative Works.  You may modify or create derivative works of the Licensed Materials (“Derivatives”).  Any Derivatives created shall have the same restrictions and limitations as listed in Sections 1.2, 1.3, 1.4 and 4.1 herein. Your use of Derivatives outside the scope of this Agreement shall require a separate license agreement.  Derivatives do not include LG Content, or other content or data bundled with, created by or otherwise displayed as a result of using the Licensed Materials (“Simulator Content”).

While it's difficult to prevent individual users from creating or using a derivative work for their own personal and private use, you would not be able to publish or otherwise share any such derivative work (whether in binary or source form) if it was based on any SVL Simulator source code.

Hope this helps.

manone888, 2021-06-27:
Thank you @lemketron and @EricBoiseLGSVL for your replies. It is good to know about the license issue. Thanks a lot. It is very helpful.