lgsvl/simulator #1519: The apollo-5.0  can't show the other vehicles  When I run the lgsvl simulator (version 2021.01.01))

luogyu7, 2021-06-27:
The apollo platform 5.0  with lgsvl  2021.0101 version could normal  running . As the following pictures:
[image]
[image]

And I can see the screen scene synchronization .  As the following picture:
[image]

I got the result picture with the following  cammands  step by step.
1. pull the docker image with the docker command:  "docker pull lgsvl/apollo-5.0";
2. "git clone --recurse-submodules https://github.com/lgsvl/apollo-5.0.git"
3. "docker/scripts/dev_start.sh";
4. "docker/scripts/dev_into.sh";
5. "./apollo.sh build_opt_gpu";
6. I run "exit" command exit the docker container and run "docker/scripts/dev_into.sh"  reenter the container;
7. "bootstrap.sh";
8. "bridge.sh";

SO, what  wrong is I haven't realize? 

could any  lgsvl&apollo experts help me?

Many thanks .

EricBoiseLGSVL, 2021-06-28:
@cyberphysicalrobotics @revati-naik ?

lemketron, 2021-06-29:
What is the problem you are having?  Looks like you're doing great so far. 

Continue on with the Apollo 5.0 instructions, from where it says "Create a simulation" here: https://www.svlsimulator.com/docs/system-under-test/apollo5-0-instructions/#launching-apollo-alongide-the-simulator

Then continue with "Open Apollo 5.0 Dreamview in a browser" and "Select a destination by clicking on a lane-line and clicking Submit Route" and finally "Watch the vehicle navigate to the destination."

EricBoiseLGSVL, 2021-06-29:
I think they are missing ground truth in dreamview, right @luogyu7 ?

lemketron, 2021-06-29:
@luogyu7 Can you show the Module Controller view from Dreamview?

luogyu7, 2021-06-29:
Very thanks for your reply.
When I run  the vehicles at the Dreamview with the ip localhost:8888 . I can get the init  
web page  with the message " please send the car init position and map data. "
[image]

and  after I run the commands as following:
1. at the "apollo" folder run :" ./scripts/bootstrap.sh";
2. input the address "localhost:8888"  (  when I input "localhost:9090"  ,It didn't work. I don't know the reason.)
3. run the command "./scripts/bridge.sh",I can see the terminal like the following picture:
[image]

4. then, I run the simulator client (version 2021.01.01),and  select a simulator I have set the configuation.
[image]

And the left column  is the situation after I click the button like the  letter "i":
[image]

the message is showling in the picture:
[image]

5. now  the situation of the simulator and Dreamview is like:
[image]

[image]
[image]

The above pictures show the “Module Controller view”  by move the simulator，And move the car via the simulator  and show the result at the dreamview(localhost:8888).
I will upload  a  short video by a link to express the situation clearly.

Here ,I thanks the reply of @EricBoiseLGSVL  . I will learn the tutorial of https://www.svlsimulator.com/docs/system-under-test/apollo5-0-instructions/#launching-apollo-alongide-the-simulator . I may not completely learn the second  half of the tutorial.  But ,I not sure if I will found the answers .But I will learn it carefully.
  
Thanks @lemketron 's reply . The Module Controller view  like following picture:
[image]

except it I add other view picture at the following :
[image]
[image]
[image]
[image]

About the “ground truth” ，where should I set at the web page dreamview  to found the ground truth?

This reply have so many pictures , many thanks for your patience.
Thanks a lot.

luogyu7, 2021-06-29:
Maybe I  have many error to debug.
I have so much content to learn.

And  about the "Select a destination by clicking on a lane-line and clicking Submit Route"
I have select the initial point and the destination point on the map  like the following picture:
[image]
After I click the "Send routing request" button .  The dreaview show a red line from the intial position to  the destination point.  BUT the car can't move automaticlly.
[image]

Where  is the worry  that I haven‘t  realized ?
Thank you very much.

EricBoiseLGSVL, 2021-06-29:
Planning should show a green line if everything is correct.  Any ideas @lemketron ?

lemketron, 2021-06-29:
> About the “ground truth” ，where should I set at the web page dreamview to found the ground truth?

The information about other vehicles in the scene either comes from the simulator (ground truth) or from the output of Apollo's perception module. You are running apollo 5.0 with perception disabled which would explain why you're not seeing the other vehicles in the scene.

Do you need to run apollo 5.0 for some specific reason? I would suggest running newer Apollo as documented here:
https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/

Running perception (and traffic light perception) requires a lot of CPU/GPU resources and makes it difficult to run on a single computer with the simulator. For that case I recommend you check out the "modular testing" documentation which shows how to use the "3D ground truth" and "signal" sensors in the simulator to provide the output that those Apollo modules would have generated.

lemketron, 2021-06-29:
> Planning should show a green line if everything is correct. Any ideas @lemketron ?

When first trying to set a destination, I recommend clicking "remove all points" and then click once somewhere across the intersection. You should have one red dot there and you can zoom in to see it using the mouse scroll wheel. Then click "Send routing request" and you should see a red path to the destination. There are other paths that also need to appear, like the outputs of routing and planning modules.

This video from the [modular testing documentation](https://www.svlsimulator.com/docs/tutorials/modular-testing/) shows a pretty good overview of what this all looks like when it's working properly: https://youtu.be/7JpxbXf8oZM

luogyu7, 2021-07-01:
Thanks for your reply and your patience.
I will try the link tutorial and research it carefully. 
Currently I'm not very clear about the   technical detail. If I have new issues to produce I will show it .
Many thanks.

closed 2021-07-02 by EricBoiseLGSVL

luogyu7, 2021-07-05:
> The information about other vehicles in the scene either comes from the simulator (ground truth) or from the output of Apollo's perception module. You are running apollo 5.0 with perception disabled [...]

I just realized the significance of this reply.  Perception  of  apollo  is a must for me. I need to test the  algorithms in perception.   This is an very important need for me.  SO, I may run perception module in my workstation（display card 2080TI x 1 ， memory:30G ）  .  Are there any methods  to run apollo's perception module?   It means a lot to me. 
Hope for your reply.
Thanks a million.

lemketron, 2021-07-06:
> I just realized the significance of this reply. Perception of apollo is a must for me. I need to test the algorithms in perception. This is an very important need for me. SO, I may run perception [...]

You can run perception in Apollo 5.0 (though 5.0 is very old) or you can try running perception in latest Apollo (master). If you have problems with perception in Apollo you might ask for help on the [Apollo issues](https://github.com/ApolloAuto/apollo/issues) page.

luogyu7, 2021-07-07:
Thanks very much.