lgsvl/simulator #1521: Adding an HD map for Apollo does not work if I follow the docs

krtt, 2021-06-27:
The docs [here](https://www.svlsimulator.com/docs/system-under-test/apollo5-0-instructions/#adding-an-hd-map) seem to be wrong or I'm doing something wrong. If I follow this guide, the step inside the Docker container is not valid. There is no script named `generate_map.sh` in the directory /apollo. All scripts seem to be in /apollo/scripts. But not with this name.

The downloaded apollo50 HD map file `base_map.bin` I already have placed in my new map dir. But to generate the other 2 needed map files, then when I searched inside the directory /apollo/scripts there is only one seemingly relevant script - named `generate_routing_topo_graph.sh`. But is it correct? And is it the only script needed or I need something more?

EricBoiseLGSVL, 2021-06-28:
@cyberphysicalrobotics @hadiTab ?

lemketron, 2021-06-29:
@EricBoiseLGSVL I think @revati-naik knows the latest on this...

AIasd, 2021-07-06:
I am encountering similar issue here. I followed [the instruction for Add an HD Map to Apollo](https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/#adding-an-hd-map). In particular, I tried "In our Apollo 5.0 fork, we have a script generate_map.sh to help users generate map files. You can copy that script to APOLLO_ROOT/scripts/ folder to use it. For details, please follow here." by copying this script to apollo/scripts/ and run it. However, I encountered the following error:

```
./scripts/generate_map.sh: line 13: bazel-bin/modules/map/tools/sim_map_generator: No such file or directory
scripts/generate_routing_topo_graph.sh: line 25: /apollo/bazel-bin/modules/routing/topo_creator/topo_creator: No such file or directory
```

AIasd, 2021-07-06:
I figured it out. The method of using generate_map.sh from 5.0 fork works but I need to do it in the docker.

revati-naik, 2021-07-07:
@krtt @Alasd For generating the map files, the first step is to place your map file (base_map.bin) in a folder under modules/map/data. Next, run your docker container. Inside your container, go to /apollo directory, and run the script generate_map.sh under /apollo/scripts (https://github.com/lgsvl/apollo-5.0/blob/simulator/scripts/generate_map.sh). This will create the map files. Please use the command `generate_map.sh MAP_FOLDER_NAME` to generate the map files inside the docker container. 

If the file generate_map.sh does not exist (in case of Apollo 6.0 or Apollo master), please create this file and paste the code from [here](https://github.com/lgsvl/apollo-5.0/blob/simulator/scripts/generate_map.sh). Also, make sure the base_map.bin is not renamed.

albert8202, 2021-11-03:
> @krtt @alasd For generating the map files, the first step is to place your map file (base_map.bin) in a folder under modules/map/data. Next, run your docker container. Inside your container, go to [...]

Why it turns to be "Permission denied", and there is no folder named bazel-bin/ in my root directory?

EricBoiseLGSVL, 2021-11-03:
This would be best asked in the apollo issues page, [here](https://github.com/ApolloAuto/apollo/issues).  I looked through and it seems there are often permission denied issues.  I recommend including the stack trace.  We will keep this open until you can successfully create the map file.

albert8202, 2021-11-11:
Thanks a lot! I ran the command 'sudo chmod -R 777 file-name' to grant the permission. I also found the bazel-bin/ directory under .cache/, and I revised the corresponding in generate_map.sh. After that, I successfully generate maps.

> This would be best asked in the apollo issues page, [here](https://github.com/ApolloAuto/apollo/issues). I looked through and it seems there are often permission denied issues. I recommend including the stack trace. We will keep this open until you can successfully create the map file.