lgsvl/simulator #1524: Perfect cameras are distorted

MaxandreOgeret, 2021-06-28:
Hello, 
I am using ROS and I am trying to overlay a lidar pointcloud over a camera and it seems that perfect cameras are still distorted.

**Hypothesis** : If the simulated camera is located at the same place as the simulated lidar. And the simulated lidar -> simulated camera Tf is good. Then the lidar points should overlay perfectly on the the image.

**Test result** : Points and images don't overlay properly.

**Meaning** : The camera images are distorted. (At least relative to the point cloud).

[image]

TFs is ROS : 

[image]

TFs in SVL : 
[image]

What do you think of this ? Is this expected behavior ?

Thanks !

MaxandreOgeret, 2021-06-28:
Update, when fiddling with the fov (then fx and fy) I find that those values work : (~~striked~~ =Config in SVL web interface, next one is a fov value that works for the overlay in ros)

(right camera)
fovx: ~~43~~ 34
fovy: ~~32~~ 25

[image]

(left camera)
fovx = ~~32.8~~ 43
foxy = ~~24.83~~ 30

[image]

---

Here's how I calculate the ideal camera info model : 

```python
    def compute_camera_info(self):
        """
        Computes the camera info message for a perfect camera with the focals and image dimensions
        """
        self.cam_info.header.frame_id = self.cam_id
        self.cam_info.distortion_model = "plumb_bob"
        self.cam_info.height = self.resy
        self.cam_info.width = self.resx

        self.fx = self.resx / (2. * tan(radians(self.fovx) * .5))
        self.fy = self.resy / (2. * tan(radians(self.fovy) * .5))

        self.cam_info.D = [0., 0., 0., 0., 0.]

        self.cam_info.K = [self.fx, 0.0,        self.resx / 2.,
                           0.0,     self.fy,    self.resy / 2.,
                           0.0,     0.0,        1.0]

        self.cam_info.P = [self.fx, 0.0,        self.resx / 2., 0.0,
                           0.0,     self.fy,    self.resy / 2., 0.0,
                           0.0,     0.0,        1.0,            0.0]

        self.cam_info.R = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
``` 

You can see tha the line for the calculation of `self.fx` and `self.fy` match your FAQ [here](https://www.svlsimulator.com/docs/support/faq/#how-do-i-get-parameters-in-camera-matrix)

**Note: You say that fx and fy must be equal. It's only true if the image would be a square. fx is linked to the x fov and fy to the y fov. If the image is not square fovx and fovx are different. Hence diffenrent fx and fy**

---

So there's clearly a discrepancy here, what could explain it ? 

Thanks !

EricBoiseLGSVL, 2021-06-28:
Will this be an issue with the cubemap lidar @kamkolak-lge ?

kamkolak-lge, 2021-07-30:
Hey @MaxandreOgeret , sorry for late reply.

I went through the math and confirmed that image posted by Simulator does not have any additional distortion and uses declared field of view. I also confirmed that using pinhole camera model with calculations provided in documentation maps points published by lidar sensor properly.

> You say that fx and fy are equal. It's only true if the image is a square.

That's not true. Documentation is correct - `fx` and `fy` are indeed equal, as this camera model has the same focal length in both axes. If it was not the case, image would appear stretched.

If we assume your parameters (width = 1444, height = 1080, fov = 60), intrinsics matrix `A` for the camera in the form of:
```
fx     0    cx
 0    fy    cy
 0     0     1
```
according to the [documentation](https://www.svlsimulator.com/docs/support/faq/#how-do-i-get-parameters-in-camera-matrix), should have the values:
```
935.3        0     722
    0    935.3     540
    0        0       1
```    

Then, given point in world space `Pw` and camera's world to local matrix (`V`):
```
// Move point from world space to camera space
Pc = V * Pw

// Apply pinhole model to move to scaled 2D coordinates
Ps = A * Pc

// Remove scaling, so that last component is equal to 1
Pi = Ps / Ps.z
```

`Pi` are pixel coordinates on the image.

Note that if lidar points are in sensor's local space, and color camera uses the exact same position and rotation, you can skip the world space altogether, as points will be basically already in camera's local space.

I took the points published by lidar, applied the method described above, and overlayed it on top of the published camera image. As you can see, they match as expected:
[image]